lcgamboa/picsimlab #2: Setting Port Directions / Possible Banksel Issue

caroper, 2018-10-29:
This code should show D0 and B7 Lit on Board 4 - PICGenios
;Start of the main program
;test dir
;
;PortD = 0b00000001          ; Initial LED Pattern
  movlw 1
  movwf PORTD
;PortB = 0b10000000          ; Initial LED Pattern
  movlw 128
  movwf PORTB
;Dir PortD Out               ;  All PortD Pins as Outputs
  banksel TRISD
  clrf  TRISD
;Dir PortB Out               ;  All PortB Pins as Outputs
  clrf  TRISB

However, when clrf TRISB executes all of the PortD LED's light.

Regards,
Chris Roper

caroper, 2018-10-29:
This Code Works in PICSimLab:

;Start of the main program
;test dir
;
;PortD = 0b00000001          ; Initial LED Pattern
	movlw	1
	movwf	PORTD
;Dir PortD Out               ;  All PortD Pins as Outputs
	banksel	TRISD
	clrf	TRISD
;PortB = 0b10000000          ; Initial LED Pattern
	movlw	128
	banksel	PORTB
	movwf	PORTB
;Dir PortB Out               ;  All PortB Pins as Outputs
	banksel	TRISB
	clrf	TRISB

Both versions work on physical hardware.

Regards,
Chris Roper

lcgamboa, 2018-10-29:
Probably the problem is in picsim and not in picsimlab. What microcontroller you used in the simulation, PIC16FXXX or PIC18FXXX?

caroper, 2018-10-29:
I tried both, but the most extensive testing was on a 16f877a as I had that one on a Dev Board for comparison to the sim. 

I am very impressed with PICSimLab, excellent work, and I was hoping to use it to develop several beginner lessons using Great Cow BASIC as the compiler. The two together would make an excellent  way for beginners, particularly in third world countries, to try out Microcontrollers without having to invest in expensive equipment and compilers. I am South African and the cost is the major barrier in most schools and for hobbyists.

Regards
Chris

lcgamboa, 2018-10-30:
Hello Chris,

this problem is really a bug in picsim in relation to the latch of the ports. Configuring the value of TRIS before using the PORT  everything works without problems, putting the value in LATCH and configuring TRIS as an output don't work in picsim as it should. I'll try to fix the problem in picsim. I will close the topic because the problem is in another project.

Thank you for this return.

closed 2018-10-30 by lcgamboa

caroper, 2018-10-30:
Thanks,
Appreciated

lcgamboa, 2018-10-30:
The bug was fixed in picsim project: https://github.com/lcgamboa/picsim/commit/2b60c7f1d848e96d6188737b0ece3b5a550e528c

caroper, 2018-10-30:
> The bug was fixed in picsim project: lcgamboa/picsim@2b60c7f

Thanks,

I found another issue related to I2C but thought it may be a related issue
so have been using software I2C in the interim.
Is there a way I can update PICSimLab or do I need to wait for a New
general release?

Thanks for taking the time to address the issue.

Regards
Chris

lcgamboa, 2018-10-30:
Chris,

try the test version https://www.dropbox.com/sh/rz1fm8b8zev5jzz/AABrk_D24Z24gkc9wNeAAeraa?dl=0
Any problem write to me.

Best regards.

Luis Claudio Gambôa Lopes

caroper, 2018-10-30:
It works

Out of interest here is the code used:
'
'  The example program below was written with the Open Source
'  Great Cow BASIC compiler (http://www.gcbasic.com/)
'
'  it Rotates the LED's on Port B and D
'  of the PICSimLab Board 4 - PICGenios.
'  (https://sourceforge.net/projects/picsim/)
'
'  Program example: Demo2c_Both_LED_Ports.gcb
'
'  Board:    PICSimLab Board 4 - PICGenios
'  MCU:       PIC16F877a
'  Clock:     8MHz
'  Compiler:  GCBASIC - (http://www.gcbasic.com/)
'  Author:    C.A.Roper - 2018
'
Config:

  #Chip 16f877a, 16           ; Target Device and Clock Speed
  #option Explicit            ; Variables must be pre declared
  '
  PortB = 0b10000000          ; Initial LED Pattern
  PortD = 0b00000001          ; Initial LED Pattern

  Dir PortB Out               ; All PortB Pins as Outputs
  Dir PortD Out               ; All PortD Pins as Outputs

Main:

  Do
    wait 500 ms
    Rotate PortD Left Simple
    Rotate PortB Right Simple
  Loop

End

It is nearly midnight here but I look forward to exploring more in the
morning.

Regards and thanks again,
Chris