lcgamboa/picsimlab #8: Serial monitor is empty on sketch which writes to serial

TomasHubelbauer, 2019-08-22:
Hey, this is in follow up on buserror/simavr#345. I have this sketch:

https://github.com/TomasHubelbauer/arduino-nmea-checksum

The repository documents how to compile it to HEX and as you can see in the sketch source code, it writes into serial. When I open its HEX it picsimlab and choose the board 5 (Arduino Uno) and open the serial viewer, I don't see any output. The port selector is empty, so I cannot select one and click Open either. Is this a problem with my sketch or the order of actions I took in picsimlab?

TomasHubelbauer, 2019-09-13:
BTW the serial doesn't work even with one of the examples you have included, the one with the pot controlling the PWM of a pin.

lcgamboa, 2019-09-14:
Have you  made the serial configuration of com0com as described in manual? 
https://lcgamboa.github.io/picsimlab/node16.html

TomasHubelbauer, 2019-09-14:
I missed that, now after installing it and going to File > Configure > COM1 and opening the Serial Terminal and selecting COM2 there and connecting, when I load my HEX file, I see the expected output!!! Thank you :-)

closed 2019-09-14 by TomasHubelbauer