lcgamboa/picsimlab #10: Build error

Tamas-F, 2020-04-01:
During build I get the following error:
```
boards/board_avr.cc:137:28: error: invalid conversion from ‘char’ to ‘unsigned char*’ [-fpermissive]
      pins[p].port = pname[1];
                     ~~~~~~~^
```
To me it seems that some conversion is missing, but couldn't figure out what would be right to do.

Also this error pops up:
```
boards/board_avr.cc:140:48: error: invalid operands of types ‘int’ and ‘unsigned char*’ to binary ‘operator|’
      avr_irq_t* stateIrq = avr_io_getirq (avr, AVR_IOCTL_IOPORT_GETIRQ (pins[p].port), pins[p].pord);
                                                ^
```
Any ideas?

lcgamboa, 2020-04-01:
Sorry. I'm working on modifying picsim and forgot to update picsimlab. The github code is already updated.

Tamas-F, 2020-04-02:
Thanks for the quick help!

closed 2020-04-02 by Tamas-F