lcgamboa/picsimlab #14: Problema com exemplos

alankardek2, 2020-07-01:
Prezado colega,

Quando abro algum projeto de exemplo do PICsimLab no MPLab X IDE v5.35 surgem alguns erros na hora de compilar. Estou usando o compilador XC8 v2.10.

Os dois primeiros erros eu consegui resolver substituindo 

#asm
	NOP
	NOP
	NOP
	NOP
	NOP
    #endasm

por 

asm("NOP");
asm("NOP");
...
asm("NOP");

Os outros problemas eu não consegui resolver.

Poderia me ajudar?

Parabéns pelo trabalho!

Seguem as telas de saída:
1) antes da correção dos dois primeiros erros:

CLEAN SUCCESSFUL (total time: 108ms)
make -f nbproject/Makefile-16F877A.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
make  -f nbproject/Makefile-16F877A.mk dist/16F877A/production/test_b3.X.production.hex
make[2]: Entering directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/display7s.p1 display7s.c 
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/eeprom.p1 eeprom.c 
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/atraso.p1 atraso.c 
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/adc.p1 adc.c 
::: advisory: (2049) C99 compliant libraries are currently not available for baseline or mid-range devices, or for enhanced mid-range devices using a reentrant stack; using C90 libraries
::: advisory: (2049) C99 compliant libraries are currently not available for baseline or mid-range devices, or for enhanced mid-range devices using a reentrant stack; using C90 libraries
::: advisory: (2049) C99 compliant libraries are currently not available for baseline or mid-range devices, or for enhanced mid-range devices using a reentrant stack; using C90 libraries
::: advisory: (2049) C99 compliant libraries are currently not available for baseline or mid-range devices, or for enhanced mid-range devices using a reentrant stack; using C90 libraries
atraso.c:36:7: error: invalid preprocessing directive
     #asm
      ^
atraso.c:42:6: error: invalid preprocessing directive
    #endasm;
     ^
2 errors generated.
(908) exit status = 1
make[2]: *** [nbproject/Makefile-16F877A.mk:190: build/16F877A/production/atraso.p1] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
make[1]: Leaving directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
make[1]: *** [nbproject/Makefile-16F877A.mk:91: .build-conf] Error 2
make: *** [nbproject/Makefile-impl.mk:39: .build-impl] Error 2

BUILD FAILED (exit value 2, total time: 17s)

2) após correção dos dois primeiros erros:

CLEAN SUCCESSFUL (total time: 235ms)
make -f nbproject/Makefile-16F877A.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
make  -f nbproject/Makefile-16F877A.mk dist/16F877A/production/test_b3.X.production.hex
make[2]: Entering directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/display7s.p1 display7s.c 
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/atraso.p1 atraso.c 
"C:\Program Files (x86)\Microchip\xc8\v2.10\bin\xc8-cc.exe"  -mcpu=16F877A -c    -fno-short-double -fno-short-float -fasmfile -maddrqual=ignore -xassembler-with-cpp -mwarn=0 -Wa,-a -DXPRJ_16F877A=16F877A  -msummary=-psect,-class,+mem,-hex,-file  -ginhx032 -Wl,--data-init -mno-keep-startup -mno-osccal -mno-resetbits -mno-save-resetbits -mno-download -mno-stackcall   -std=c99 -gdwarf-3 -mstack=compiled:auto:auto     -o build/16F877A/production/eeprom.p1 eeprom.c 
[... 16 lines cut ...]
::: advisory: (2049) C99 compliant libraries are currently not available for baseline or mid-range devices, or for enhanced mid-range devices using a reentrant stack; using C90 libraries
teste_b3.c:66:6: error: variable has incomplete type 'void'
void interrupt isrh(void);
     ^
teste_b3.c:66:15: error: expected ';' after top level declarator
void interrupt isrh(void);
              ^
              ;
teste_b3.c:542:6: error: variable has incomplete type 'void'
void interrupt isrh() 
     ^
teste_b3.c:542:15: error: expected ';' after top level declarator
void interrupt isrh() 
              ^
              ;
4 errors generated.
(908) exit status = 1
make[2]: *** [nbproject/Makefile-16F877A.mk:254: build/16F877A/production/teste_b3.p1] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [nbproject/Makefile-16F877A.mk:91: .build-conf] Error 2
make: *** [nbproject/Makefile-impl.mk:39: .build-impl] Error 2
make[2]: Leaving directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'
make[1]: Leaving directory 'C:/Users/alank/Downloads/picsimlab-master/docs/examples/board_3/PIC16F877A/test_b3/src/test_b3.X'

BUILD FAILED (exit value 2, total time: 20s)

lcgamboa, 2020-07-01:
Olá Alan,

As partes em assembly dos códigos dos exemplos não são compatível com o c99. 
Nas novas versões do MPLABX ele troca a versão do projeto para c99 automáticamente ao abrir. Vá nas propriedades do projeto e escolha o padrão c90 nas opções do xc8 que vai compilar sem erro. O problema é o MPLABX que não respeita as configurações dos projetos de versões anteriores.

Abraço

closed 2020-07-02 by lcgamboa

alankardek2, 2020-07-02:
Olá Luís Claudio,

Funcionou! :)
Muito obrigado! Realizei o seguinte procedimento:

Abra o MPLab X IDE, clique em File, Open Project... e selecione a pasta onde descompactou o arquivo de exemplo, test_b4.X, e clique em Open Project.
Aceite a mensagem de upgrade do projeto.
Torne a pasta do projeto como o projeto principal (botão direito do mouse em cima da pasta text_b4.X e clique em Set as Main Project).
Antes de compilar o projeto, clique o botão direito do mouse na pasta do projeto e escolha Set Configuration, PIC16F877A.
Clique novamente o botão direito do mouse na pasta do projeto e escolha Set Configuration, Customize...
Clique na opção "XC8 Global Optionis", dentro de "Conf:[16F877A]" e altere o padrão C 99 para C 90.