lcgamboa/picsimlab #17: second usart support PIC18F27K40

simoelk, 2020-08-08:
https://github.com/lcgamboa/picsimlab/blob/355f26d11d0674b9fe3b6a78dc79be79a89be7cd/src/boards/board_picsim.cc#L35

I think you should set the name of serial port also for the second usart ! 

 pic_set_serial (&pic,1, port, 0, 0, 0);

PS: i have tested this solution on the PIC18F27K40 in the Curiosity HPC board .

Best regards,

lcgamboa, 2020-08-08:
I have added a combo to support second serial port selection on commit 258542b84771bb5ae73dae2adec6eb34cf2238ed .
Now you can choose the second port.
Please test if it's working, I don't have code ready to test the PIC18F27K40 serials.
You can use the script with "exp" option to compile experimental support `picsimlab_build_all_and_deps.sh exp`, it takes a little longer to compile because of qemu_stm32.

[image]

simoelk, 2020-08-08:
Yes , it works ! thank you.  :100:

closed 2020-08-08 by simoelk

lcgamboa, 2020-08-19:
Hi Mohammed,
I intend to release picsimlab release 0.8.2 soon. Do you have any consideration for your new board?

Best regards,

 Luis Claudio Gambôa Lopes 

 
 

https://github.com/lcgamboa/picsimlab/blob/355f26d11d0674b9fe3b6a78dc79be79a89be7cd/src/boards/board_picsim.cc#L35

simoelk, 2020-08-19:
Hi M.Gambôa,

No, currently with the tests that i have done i think it's working fine.

Sincerely,

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>