lcgamboa/picsimlab #18: trigger issues

nerdralph, 2020-09-21:
I'm using 0.8.2, simulating a UNO, and the oscilloscope triggers don't seem to work.  It triggers even when the level stays at 0V.
[image]

lcgamboa, 2020-09-21:
Hi, in single mode (single toggle button pressed) after first trigger the capture stop (stop toggle button pressed). Press stop to acquire again. Press single button to return to normal mode. 
Try with this example https://lcgamboa.github.io/picsimlab_examples/examples/examples_index.html#board_Arduino_Uno_atmega328p_AnalogInOutSerial
If the examples not work send me more info about your code.

nerdralph, 2020-09-22:
I'm familiar with how oscilloscope triggering normally works; I have a Rigol DS1054Z.
It appears you are using simavr, so I'll give that a try using it's VCD feature.  It supports more than just the ATmega328, which would be more useful for me.

lcgamboa, 2020-09-22:
I'm sorry, but only the information "simulating a UNO, and the oscilloscope triggers don't seem to work." it is not enough for me to be able to identify what the problem is. 
The PICSimLab oscilloscope is very simple and limited, it cannot be compared to a real oscilloscope.
PICSimLab only supports ATMEGA328P (and ATMEGA2560 in experimental version) using simavr. The original simavr does not have bitbang signals for TWI, SPI and UART that are supported in the simavr used in PICSimLab. You can generate VCD signals using the VCD_DUMP part, look at the example: https://lcgamboa.github.io/picsimlab_examples/examples/examples_index.html#board_Arduino_Uno_atmega328p_Serial_LCD 
If you can send more information about the code you are using it would be easier to identify the problem.

closed 2020-09-30 by lcgamboa