lcgamboa/picsimlab #21: ETH w5500 problem

paoloSantinelli, 2020-11-01:
Good Morning,

I have downloaded picsimlab and compiled (cross compiled the win version in to Ubuntu) the experimental version to try the new spare part ETH w5500.

1) I have attached an Arduino Uno to w5500 and have run the telent client example. The windows machine running picsimlab has IP: 192.168.1.22, while I have statically assigned the IP 192.168.1.240 to the w5500 (I have even tried with 192.168.1.22);
2) On a machine (192.168.2.214) on the same network I have used netcat listening on port 2000 for incoming TCP connection.
echo "Hello" | nc -l 2000

3) When Arduino try to connect, the connection is established but it is immediately closed (RST) by picsimlab;

This is what I have the captured using wireshark about the network activity:

- IP Win machine running picsimlab = 192.168.1.22
- IP Ubuntu machine runnung netcat = 192.168.2.214
-------------------------------------------------------------------
1	192.168.1.22	192.168.1.214	TCP	66	51004 → 2000 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM=1

2	192.168.1.214	192.168.1.22	TCP	66	2000 → 51004 [SYN, ACK] Seq=0 Ack=1 Win=64240 Len=0 MSS=1460 SACK_PERM=1 WS=1024

3	192.168.1.22	192.168.1.214	TCP	54	51004 → 2000 [ACK] Seq=1 Ack=1 Win=2102272 Len=0

4	192.168.1.214	192.168.1.22	TCP	60	2000 → 51004 [PSH, ACK] Seq=1 Ack=1 Win=64512 Len=6

5	192.168.1.22	192.168.1.214	TCP	54	51004 → 2000 [RST, ACK] Seq=1 Ack=7 Win=0 Len=0
---------------------------------------------------------------------

This is the message on the Arduino Serial Terminal:

connecting...
connection failed

disconnecting.

BTW. 
The w5500 web server example seem to work perfectly!

Any help will be appreciated.

Thank you very much

lcgamboa, 2020-11-01:
Hi Paolo,

I'm sorry, I only tested the ETH w5500 on Linux. The ETH part w5500 is not yet in the experimental stage, it is in the development stage yet (which is done in Linux). I develop everything myself and normally I don't have time to test everything, any help is welcome.
I fixed the problems in relation to sockets in non block mode on Windows (93a53bb0ce7cc9185ac98fbb3c784d2ec46f9a71), but I haven't tested all the examples. In Linux all examples from the Arduino Ethernet library work.
Your simulation should now work on windows. Any problem just let me know.

paoloSantinelli, 2020-11-01:
Hi 

you don't have to apologize! Thank you very much for your great work!

I am going to test the code now. 

Thank you again!

paoloSantinelli, 2020-11-01:
Sorry,

it seems the problem is still there. 

But I am new to github so.... I have removed the old cloned pixsimlab folder from my linux machine, then I have cloned and build again:

git clone https://github.com/lcgamboa/picsimlab.git
cd picsimlab
./picsimlab_build_w64.sh exp

Did I do some mistakes... Maybe I miss something ??!!

Thanks and Sorry again

Paolo

lcgamboa, 2020-11-01:
Apparently you did everything right.
I tested the corrections in Linux using wine, I will actually test in windows to check.

paoloSantinelli, 2020-11-02:
Thank you again

lcgamboa, 2020-11-02:
I run the PICSimLab on windows and fixed the bugs on TCP connection. I have tested three examples and all working now. Test if your code work with the last fix.

paoloSantinelli, 2020-11-02:
Hi,

Yes! I have just tested, it seems to work perfectly!!

I tested it on the same machine (Arduino client with ETH w5500 and netcat server on the same VM) and even with netcat server running on a different machine.

Thank you Again!!

closed 2020-11-03 by paoloSantinelli