lcgamboa/picsimlab #22: Problema com uso simultâneo de Aquecedor e Ventilador - PICGenios PIC18F4520

alankardek2, 2020-11-23:
Prezado Luis Claudio Gambôa,

Estou tentando fazer funcionar o aquecedor e o ventilador na placa PICGenios usando o microcontrolador PIC18F4520.
Quando não aciono o ventilador via PWM, o aquecedor funciona. 
Quando rodo a função PWM1_Start(), o aquecedor para de funcionar. Mais especificamente quando o registrador T2CONbits.TMR2ON é ativado, dentro de pwm.c.
Achei que pudesse ser alguma configuração de RC5 no modo SPI. Inclui SSPCON1bits.SSPEN = 0, mas sem sucesso.

Poderia me ajudar?

Seguem mais detalhes do código:

==============================================================
#define _XTAL_FREQ 8000000
#include "configbits.h"
#include <xc.h>
#include "flexlcd.h"
#include "adc.h"
#include <stdio.h>
#include "itoa.h"
#include "pwm.h"
#include "serial.h"

//Variables
unsigned int duty = 100, infrav = 0, count = 0, result = 0; // 0 a 255
char str[5], buffer_tx[20], buffer_rx[20];
// resistência de aquecimento RC5
#define aquec PORTCbits.RC5

void main(void) 
{
    ADCON0 = 0b00000000; // Configura todos pinos para digital
    ADCON1 = 0x00; // habilita todas as portas analógicas
    INTCON2bits.RBPU = 1; // Pull-up resistors off
    T1CON = 0b10000011; // configura Timer1
    TMR1L = 0; 
    TMR1H = 0; // valor inicial

    // Input or Output
    TRISA = 0xFF;
    TRISB = 0b00000000; 
    TRISC = 0b10000001;
    TRISD = 0b00000000; 
    TRISE = 0b00000001;

    // Clear ports
    PORTA = 0; 
    LATA = 0;
    PORTB = 0; 
    LATB = 0;
    PORTC = 0; 
    LATC = 0;
    PORTD = 0; 
    LATD = 0;
    PORTE = 0; 
    LATE = 0;    

    aquec = 1;  // aquecedor ligado
    
    serial_init();
    adc_init(); // configura registradores ADCON
    Lcd_Init();
    Lcd_Cmd(LCD_CURSOR_OFF); 
    PWM1_Init(1000);  // configura frequencia do PWM para 1kHz
    PWM1_Start(); // configura registradores e inicia PWM
    //Timer 0 config
    INTCONbits.TMR0IF = 0; // TMR0 register has overflowed (must be cleared in software)
    INTCONbits.TMR0IE = 1; //1 = Enables the TMR0 overflow interrupt
    T0CON = 0b11000101; //  8bit and 1:64 Prescale
    TMR0 = 131;  
    INTCON = 0b11100000; // enable gobal and peripheric interrupts
    
    while(1) 
    {
        CLRWDT();         
        serial_rx_str_until(buffer_rx,20,'\n');  
        Lcd_Out(2, 0, buffer_rx);

    }
    return;
}

void interrupt isr(void)
{
    if (INTCONbits.TMR0IF == 1)
    {
        INTCONbits.TMR0IF = 0; // Clear interrupt flag
        TMR0 = 131;
        count ++;
        if (count == 250)
        {
            count = 0;
            LATDbits.LATD1 = ~LATDbits.LATD1; // Toggle led  
            PWM1_Set_Duty(duty);        
            infrav = TMR1L;
            infrav += TMR1H<<8;
            sprintf(str, "%05u",  infrav);        
            Lcd_Out(1, 0, str);
            result = adc_amostra(1); // sensor de temperatura LM35 canal 1, 
            sprintf(buffer_tx, "/*%04d/", result);
            serial_tx_str(buffer_tx);               
        }        
    }
}

============================================================
CÓDIGO DE pwm.c

#include <xc.h>
#include"pwm.h"

 void PWM1_Init(unsigned int f)
  {
   unsigned int temp;
 //PWM Period = [(PR2) + 1] * 4 * TOSC *(TMR2 Prescale Value)
 //PWM Duty Cycle = (CCPRXL:CCPXCON<5:4>) *TOSC * (TMR2 Prescale Value)

   //desliga PWM
     CCP1CON=0x00;//CCP disabled
     TRISCbits.TRISC2=1; //desliga saídas PWM
     TRISDbits.TRISD5=1;

     PORTCbits.RC2=0; //deliga saídas PWM
     PORTDbits.RD5=0;

     CCPR1L=0;//ou 255?

     //calculo TMR2

      T2CONbits.TMR2ON=0;
             
     temp=_XTAL_FREQ/(f*4l);
    
     if (temp < 256)
     {
       T2CONbits.T2CKPS=0;  //1
       PR2=temp;
     }
     else if(temp/4 < 256 )
     {
       T2CONbits.T2CKPS=1;  //4
       PR2=(temp+2)/4;
     }
     else
     {
       PR2=(temp+8)/16;
       T2CONbits.T2CKPS=2;  //16
     }

     T2CONbits.T2OUTPS=0;  //1-16
     
  }
  void PWM1_Start(void)
  {

      TRISCbits.TRISC2=0; //liga saídas PWM
      TRISDbits.TRISD5=0;

      CCP1CON=0x0F; //CCP -> PWM mode 0x0F
            
      T2CONbits.TMR2ON=1;
      
      // tentativa de acionar LC5, sem sucesso
      SSPCON1bits.SSPEN = 0;
      TRISCbits.TRISC5=0;
      LATCbits.LC5 = 1;

      //espera PWM normalizar

      PIR1bits.TMR2IF=0;
      while(PIR1bits.TMR2IF == 0);
      PIR1bits.TMR2IF=0;
      while(PIR1bits.TMR2IF == 0);
  }

  void PWM1_Set_Duty(unsigned char d)
  {
      unsigned int temp;
      
      temp=(((unsigned long)(d))*((PR2<<2)|0x03))/255;

      CCPR1L= (0x03FC&temp)>>2;
      CCP1CON=((0x0003&temp)<<4)|0x0F;
  }

lcgamboa, 2020-11-23:
Boa tarde Alan,

Esse problema é causado por um bug no código do PICsim relacionado ao PWM, corrigido a algum tempo https://github.com/lcgamboa/picsim/commit/02c34e7e29e2f3f361b8db99ded19f5f0ddae291 .  
Estou finalizando os ajustes para o lançamento de uma nova versão do PICSimLab que já vai vir com esse problema resolvido.
Enquanto isso você pode tentar utilizar a versão compilada com correção dos links abaixo:
https://www.dropbox.com/s/62645a46cpnnerv/picsimlab_0.8.2_win64_setup.zip?dl=0
https://www.dropbox.com/s/lqecwsyrro6ynxz/picsimlab_0.8.2_win32_setup.zip?dl=0    

Abraços e desculpe o transtorno.

alankardek2, 2020-11-23:
Boa tarde Luis,

Primeiro, gostaria de te agradecer pelo suporte e pela rapidez na resposta.

Instalei a versão compilada com correção e funcionou!

Aproveito a oportunidade para perguntar se o PICSimLab também dá suporte para algum RTOS, como o uCRTOS ou o FreeRTOS. Eles funcionam com alguma placa? Pensei em rodar com a placa Arduino Uno.

Obrigado,
Alan.

lcgamboa, 2020-11-23:
Boa tarde Alan,

a muito tempo testei o brtos e o  FreeRTOS no PICSimLab utilizando o PIC18F, o exemplo do brtos funcionou bem, os exemplos do FreeRTOS  no PIC18F ficaram muito limitados (gastaram toda memória para piscar LEDs) e que me lembre tinha uns 3 exemplos e todos funcionaram. Quando testei ainda não tinha o suporte aos AVR e STM32 no PICSimLab, mas acredito que funcione também. Ná página do Qemu stm32 (simulador usado em segundo plano pelo PICSimLab) há descrição de suporte ao  FreeRTOS. Acho que utilizando o stm32 ou o atmega2560 na versão experimental do PICSimLab deve funcionar.
Devido a sobrecarga de trabalho tenho tido muito pouco tempo para desenvolver o PICSimLab e executar testes. Se você fizer alguns testes usando RTOS por favor me de um retorno.

Abraço

alankardek2, 2020-11-25:
Bom dia Luis,

Assim que conseguirmos realizar algum teste eu retorno sim. Mais uma vez, muito obrigado.

ABraço

lcgamboa, 2020-11-28:
A nova versão com a correção do problema foi lançada.

closed 2020-11-28 by lcgamboa

alankardek2, 2020-12-02:
Muito obrigado!