lcgamboa/picsimlab #23: Instabilidade da porta serial na versão 0.8.5

alankardek2, 2020-12-04:
Prezado Luis,

Quando instalei a versão 0.8.5 eu percebi instabilidade na comunicação serial.
Na versão 0.8.2 tudo rodava normalmente, quando atualizei, comecei a ter problemas. Os bytes começaram a tramitar com atraso, o que comprometeu a comunicação.

Alan.

lcgamboa, 2020-12-05:
Nesta última versão foi adicionado o suporte para a comunicação serial gerar os sinais nos pinos do microcontrolador. Logo houve modificações no código das portas seriais. 
Qual placa e microcontrolador você está utilizando?
Se for possível enviar um código do microcontrolador que funcionava na versão anteriror e consiga reproduzir o problema na nova versão ajudaria muito encontrar o problema.

lcgamboa, 2020-12-07:
O problema no PICSim foi resolvido e os arquivos para download atualizados.

closed 2020-12-07 by lcgamboa