lcgamboa/picsimlab #24: Blue Pill board

alankardek2, 2020-12-04:
Olá Luis,

Para tentar implementar o RTOS no STM32, inicialmente tentei testar um programa para enviar um caracter pela porta serial. No entanto, quando inicio a placa, a porta COM aparece com status de erro: "Serial: COM1' (ERROR)".

Teria como resolver este problema?

lcgamboa, 2020-12-05:
O STM32 roda em um programa separado, o qemu-stm32. Ele se comunica com o PICSimLab, que o controla. Na versão atual ainda não há integração sobre o status da porta serial. Logo mesmo com a mensagem de erro no PICSimLab (sempre aparece o status ERROR) a porta serial deve estar funcionando. Depois da porta escolhida no PICSimLab é necessário fechar e abrir o programa para que a alteração faça efeito.  No Linux a primeira porta serial está funcionando sem problemas, não me lembro se testei no Windows ainda. Alguns programas do STM32 utilizam outra porta serial sem ser a primeira do ARM, existe suporte no qemu-stm32, mas ainda não existe suporte no PICSimLab. Esse exemplo https://github.com/beckus/stm32_p103_demos/blob/master/demos/freertos_multithread/main.c funciona imprimindo as mensagens:

```
[22:01:33:054] Hello 2␊
[22:01:33:521] Hello 2␊
[22:01:33:989] Hello 2␊
[22:01:34:457] Hello 2␊
[22:01:34:925] Hello 1␊
[22:01:34:926] Hello 2␊
[22:01:35:393] Hello 2␊
[22:01:35:861] Hello 2␊
[22:01:36:329] Hello 2␊
[22:01:36:797] Hello 1␊
[22:01:36:798] Hello 2␊
[22:01:37:265] Hello 2␊
```

Favor me informar se não conseguir fazer funcionar no Windows.

lcgamboa, 2020-12-14:
A mensagem de erro foi removida na atualização da ultima versão.

closed 2020-12-14 by lcgamboa