lcgamboa/picsimlab #25: Arduino Uno PD0 PD1

CloudVIII, 2020-12-20:
Hi, wrote code with asm that uses pins PD0, PD1 for an ATmega328P and UART communication it's disabled, but can't use them properly in PICSimLab.

lcgamboa, 2020-12-20:
Apparently the simavr initializes the UCSR0B register with the value 0x08 (instead of 0x00), enabling the TX pin of the serial and disabling the PD1. Assigning the value 0x00 to UCSR0B at the beginning of the atmega code solves the operating problem. I will try to find the exact source of the problem.

lcgamboa, 2020-12-20:
Confirmed, the issue is in the simavr (used in picsimlab background to simulate avr mcus):
https://github.com/buserror/simavr/blob/0e03bc6f48b078606b9a970594ee0ef82ffa0117/simavr/sim/avr_uart.c#L466
The simavr force the enabling of TX pin for debug purpose.

CloudVIII, 2020-12-21:
Sadly, already tried that in the code. Still the same problem.

lcgamboa, 2020-12-21:
I have tested with this c code:

```
#define F_CPU 2000000L
#include <avr/io.h>
#include <util/delay.h>

int main()
{
    unsigned char val=0x01;
    DDRB=0xFF;
    DDRC=0xFF;
    DDRD=0xFF;
    UCSR0B=0x00; //need to work in simavr
    PORTD|=0x02;  //need to work in simavr
    PORTD&=~0x02;  //need to work in simavr
    while(1)
    {
        PORTB=val;
        PORTC=val;
        PORTD=val;
        val<<=1;
        if(!val)val=0x01;
        _delay_ms(500);
    }
}
```
Can you send me your sample code with problem?

lcgamboa, 2020-12-21:
Simavr txpin wrong state on reset fixed in commit ddc38c6

CloudVIII, 2020-12-21:
It works after the last commit.

closed 2020-12-21 by lcgamboa