lcgamboa/picsimlab #27: Is there a way to display a custom character on 16x2 LCD screen on

wvffle, 2021-01-30:


closed 2021-01-30 by wvffle

devdanzin, 2021-01-30:
@wvffle did you solve the issue? If so, can you explain what it was? Thanks!

wvffle, 2021-01-31:
Sure thing,

To display the custom character you have to first write it to the CGRAM. You can do so by using command `lcd_cmd(0x40);` (if I remember correctly).
Then you define the character. It consists of 8x 5-bit integers, bits of value `1` are turning on respective pixels. Then go back to the line one/2 with `lcd_cmd(L_L1);` and write your characters with `lcd_dat(index);` where `index` is the index of your character. As far as I understand, you can define 8 characters at most.

So to display a simple face in first line, you'd do:

```c
lcd_cmd(0x40); // you can add 8 * index if you want to define character at index

lcd_dat(0b00000);
lcd_dat(0b00000);
lcd_dat(0b01010);
lcd_dat(0b01010);
lcd_dat(0b10001);
lcd_dat(0b01110);
lcd_dat(0b00000);
lcd_dat(0b00000);

lcd_cmd(L_L1);
lcd_dat(0);
```