lcgamboa/picsimlab #28: 28Pin PIC devices Crash on Hex Load

caroper, 2021-02-04:
I have just installed the latest version of PICSimLab and was pleased to see the "BreadBoard" option and its updated Device list.
Unfortunately the first devices that I tested with it, namely the  PIC16f18855 and PIC16f1788, both crashed the simulator upon loading the HEX file.

In Both cases I was using a Simple GCBASIC source that was compiled and verified on real hardware first.
`
#chip 16f18855

Dir PortA Out

Do
  PortA = NOT PortA
  Wait 250 ms
Loop

`

[image]

The older devices that were included in the previous versions of PICSimLab appear to work in the Breadboard but not the newer 28 Pin Devices.

Regards
Chris

lcgamboa, 2021-02-05:
The problem was in picsim and it has already been fixed  https://github.com/lcgamboa/picsim/commit/18b3156ab7664e0ced34496c1c79331ba81062bd 
The 28-pin devices have PORTE registers but do not have LATE, PICsim failed to try to update LATE. 

I will update the files for download tomorrow and let you know here. 

Thanks for the collaboration.

lcgamboa, 2021-02-05:
I updated the files for download, now it should work without problems.

caroper, 2021-02-05:
Thank you for the quick response.
I look forward to continuing my tests.

closed 2021-02-10 by lcgamboa