lcgamboa/picsimlab #29: Serial Terminal problems

tombo9999, 2021-02-07:
Hi, I'm Luca from Genova Italy.
I'm going to write an article in the online magazine https://all3dp.com/ titled "X Best Arduino Simulators (Online & Offline)".
I'm testing your software and I'm having issues with the serial output.
Installed com0com on Windows 10 and it works properly (see image)
But if I load a hex file as simple as just printing output to the serial, it doesn't work.
At first, tried your example AnalogInOutSerial.pzw

I know it's surely my mistake, but can you give me some hints? Thank you in advance.
Luca Dinale

[image]

lcgamboa, 2021-02-07:
Hi Luca,

there is a limitation of windows with serial ports larger than COM9 (https://support.microsoft.com/en-us/topic/howto-specify-serial-ports-larger-than-com9-db9078a5-b7b6-bf00-240f-f749ebfd913e) , I am already correcting the PICSimLab code. In the current version it works with the ports from COM1 to COM9, sorry for the inconvenience. 

The HC-SR04 is on the TODO list, I create the components on demand, you are the first to order. I will try to implement and inform you. I believe I don't have a simple way to simulate without the component.

Thank you for your feedback and for helping to improve PICSimLab. 

Best regards,

Luis Claudio.

lcgamboa, 2021-02-07:
Hi Luca,

I made the correction to use serial ports larger than COM9 on windows and added the HC-SR04 part. I tested it with your code and it worked without problems. The github source code is already updated with the changes. In the link below the latest version compiled for windows 10 64 bits (and your example converted to PICSimLab).

https://www.dropbox.com/s/4hax3qn3k1b407s/picsimlab_latest.zip?dl=0

Best regards,

Luis Claudio.

tombo9999, 2021-02-07:
Thank you, Luis Claudio.
It works like a charm.

Thank you again for your quick response 😁

Have a nice time, Luca

tombo9999, 2021-02-08:
Just for fun, tried a quick SVG version of the part.
If you think you can use it, I can modify it as you like (I already saw a
small mistake).
Enclosed Inkscape version, plain SVG, and PNG files.

Ciao, Luca 😁

lcgamboa, 2021-02-08:
Hi Luca,

if you can send it would be a help, I intend to use svg in the future instead of png to improve the graphic quality of the simulator. I'm studying to use PcbDraw https://github.com/yaqwsx/PcbDraw to redo the design of the boards.

Best regards,

Luis Claudio.

closed 2021-02-10 by tombo9999

tombo9999, 2021-03-26:
For your info: https://all3dp.com/2/best-arduino-simulators-online-offline/

Have a nice time, Luca

lcgamboa, 2021-03-26:
Hi Luca,
congratulations, the article was very good. 

I am finalizing the new version of PICSimLab with support for figures in svg format, if you can collaborate with your hc-sr04 svg I would be grateful. The online version is already updated to use the svg board figures, but it will take a while to redo all the pictures of all parts. 
I would like to know if I can add the code for your hc-sr04 usage example in the PICSimLab examples. That way it could be accessed online, and if you want you can add the link in the article.
As for the generation of schematics for creating PCBs, it is a future plan to integrate with kicad. I am waiting for the kicad 6.0 file format change to start implementing.
As for the usability of PICSimLab, I would like to know in your opinion what can be improved to make the user's life easier. Most have difficulty configuring the virtual serial port, I got around this by creating a virtual terminal part that dispenses with the installation of the virtual serial port. (but to program directly from the Arduino IDE, you still need the virtual port). Despite many downloads, the users' feedback is very low, they usually only get in touch when there is a malfunction, I never received any feedback on the usability or interface. As you used several simulators your opinion would contribute a lot.   

Best regards,

Luis Claudio

tombo9999, 2021-03-26:
Hi Luis,
for the HC-SR04 image I used a free SVG file I found at this address:
https://freesvg.org/ultrasonic-distance-measurement-module
I think you can use it freely, pun intended 😉
And yes, the virtual serial port is not really user-friendly.
I'll give you my feedback about UI and usability in a few days.

Have a nice weekend, Luca