lcgamboa/picsimlab #31: Picsimlab not starting

almontasser, 2021-04-04:
Picsimlab not starting after I've installed MPLABX. It was running without problems before installing MPLABX.

lcgamboa, 2021-04-04:
Just installing mplabx does not affect PICSimLab in any way. 
Did you install the picsimlab_md plugin on mplabx? 
What version are you using (mplabx, PICSimLab and picsimlab_md)? Which operating system? 
I need as much information as possible so that I can try to replicate the problem to try to solve it.

almontasser, 2021-04-05:
OS: Windows 10 Pro 20H2
PICSimLab: 0.8.6
MPLABX: 5.45
com-picsim-picsimlab_mplabx5.45.nbm

PICSimLab was working fine last week, and I don't think I have installed anything, or adjusted any thing on the system, except of installing MPLABX.

I have tried restoring windows from restore point. did not work ether.

lcgamboa, 2021-04-05:
Normally PICSimLab stops working on windows when a configuration file is corrupted or an improper .hex file is loaded. 
The easiest way to solve this problem is to upload a .pzw file. Which can be downloaded from the [examples](https://lcgamboa.github.io/picsimlab_examples/examples/examples_index.html). 
Double click on the .pzw file and see if PICSimLab works.

almontasser, 2021-04-06:
I already tried reinstalling PICSimLab, and already tried to open an example that I've downloaded and worked before.

If you can give me directions on how to debug the program from source, I might be able to solve the problem.

almontasser, 2021-04-06:
I think I was able to reproduce it. I've installed it in a new virtual machine. At first it works fine, but when I changed the board to BluePill board it crashed and did not start after that.

lcgamboa, 2021-04-06:
The extra boards of the experimental version are not yet 100% free of critical problems (so they are not yet in the stable version).
The BluePill board uses the qemu-stm32 running together with PICSimLab. The communication between the two is done through TCP/IP sockets on Windows and Unix socket on Linux. 
At first, there must be some synchronization problem between the two programs, caused by communication problems due to a firewall or blocking the execution of qemu-stm32 by an antivirus program.   
Edit the picsimlab.ini file and change the board manually:
`picsimlab_lab = "PICGenios"`
And do a test to see if it works again. 
The picsimlab.ini file on my windows is in the path:  c:\users\username\AppData\Roaming\picsimlab\picsimlab.ini

lcgamboa, 2021-04-07:
I was unable to reproduce the problem on my computer. Another user reported the same problem. I made some changes so that when a .pzw file is loaded, the settings return to the default. 8e9474f

It is also possible to reset to a default state using the command (instead edit picsimlab.ini): 
`picsimlab Arduino_Uno 0`

If you get any more information or are able to solve the problem please let me know.

almontasser, 2021-04-07:
I works fine after editing picsimlab.ini, but I don't have time to figure out the exact cause, maybe later.

lcgamboa, 2021-04-07:
When you have time to run more tests, the version below has the output console active. Some relevant information about the problem may appear. 

https://www.dropbox.com/s/mzwwgplsyvggxi1/picsimlab_0.8.7_experimental_win64_setup.zip?dl=0

wilianfc, 2021-05-07:
Hi lcgamboa,

Unfortunatelly, picsimlab justo not open the graphic interface anymore since last windows 10 updates. I tryed to reinstall the app and edit .ini file and the picsimlab Arduino_Uno 0 command, but no effect. Do you have some clues about how to fix this issue?
Thanks

lcgamboa, 2021-06-19:
I found the problem that caused the crash and fixed it in the commit  b8c25a788cafe8a3edfb599f396709a6169db65f  .
The next version will already be free of the problem.

closed 2021-06-22 by lcgamboa