lcgamboa/picsimlab #32: README's link to the documentation is broken.

gatk555, 2021-04-05:
I did track them down, but it is not straightforward.

lcgamboa, 2021-04-05:
I recently split the documentation into another repo. Thanks for reporting, I already made the correction. 44eddae

gatk555, 2021-04-08:
Thanks!  A question: is it possible to use PCsimlab to create input VCD
files for input to simavr?  I am thinking about applications like
regression testing,
where a user uses an interactive program to create a firmware test and then
reruns the interaction with firmware semi-automatically after changes.

Thanks,

Giles

lcgamboa, 2021-04-08:
Yes, it's possible.
You can use the VCD play part:
Part: https://lcgamboa.github.io/picsimlab_docs/VCD_Play.html
Example: https://lcgamboa.github.io/picsimlab_examples/examples/examples_index.html#board_Arduino_Uno_atmega328p_rc_receiver
It is possible to create a VCD file of up to 8 channels. It is also possible to use more than one VCD play. 

Or you can use the remote control TCP / IP interface to control the boards or parts using a script or program. In this way it is possible, for example, to activate the keys and buttons and read the status of the output LEDs. The remote control interface was designed precisely for automation and code tests that need user intervention. 
https://lcgamboa.github.io/picsimlab_docs/Remote_Control_Interface.html

For test automation it is also possible to use the console PICSimLab version without the GUI using the remote control interface.

gatk555, 2021-04-09:
Thanks.  I will take a more detailed look at it.

closed 2021-05-03 by lcgamboa