lcgamboa/picsimlab #33: not able to install with .deb or with automatic installer for linux

sumvidh, 2021-04-12:
Dependency is not satisfiable :libwxbase3.0-0v5 (>=3.0.5.1+dfsg)

tried app image also previous version
also updated from linux mint 20.04 to 20.10 but no lu

lcgamboa, 2021-04-12:
Are you using the latest version [picsimlab_0.8.6_Ubuntu_20.10_amd64.deb](https://github.com/lcgamboa/picsimlab/releases/download/v0.8.6/picsimlab_0.8.6_Ubuntu_20.10_amd64.deb) ?

I don't have linux mint here, but the packages are compiled for ubuntu 20.10 and tested on debian where they work without problems.

sumvidh, 2021-04-12:
Hey,
Apparently it just works fine ...I mean the error is there but it shows up in menu and works fine ...but must use terminal (dpkg -i pkg )

sumvidh, 2021-04-12:
You can close the issue!!

sumvidh, 2021-04-12:
Probably can just close the problem...
Also is there way I can contribute to the project (even documentation stuff will do)

closed 2021-04-12 by sumvidh