lcgamboa/picsimlab #34: Spare Parts window does not show in Win10 - PICSimLab 0.8.6

cemonem, 2021-04-14:
Hello,

A student of mine is not able to view the spare parts window. They are supposed to load the configuration through the workspace/spare parts window but it the window only shows up in the taskbar, it is not visible. Could you please help us solve the problem? The screenshot and the configuration files are attached.

[image]

[confs.zip](https://github.com/lcgamboa/picsimlab/files/6313144/confs.zip)

Version: 0.8.6
OS: Windows 10 x64

Regards,

A. Cem Onem

lcgamboa, 2021-04-14:
Apparently your student has moved the window out of the visible area (or it was positioned on a monitor with a larger screen). Normally, clicking on the window icon on the taskbar and maximizing the window causes it to appear again. 
In the .pzw file sent by you the window is in the position: 
```
spare_position	= "2542,72,573,327"
```
Attached is the file with the corrected position for visible area. 
[workspace.pzw.zip](https://github.com/lcgamboa/picsimlab/files/6313658/workspace.pzw.zip)

cemonem, 2021-04-15:
It works now, Thank you for your help.

closed 2021-04-15 by cemonem