lcgamboa/picsimlab #35: SUBLW Instruction doesn't set the N bit of STATUS register (PIC18F4620)

necrashter, 2021-05-05:
## Expected Behavior
According to [the datasheet of PIC18F4620 (Page 305)](http://ww1.microchip.com/downloads/en/DeviceDoc/39626e.pdf), SUBLW instruction must affect the N bit of STATUS register. It should be 1 if the result is negative, and 0 otherwise.

## Current Behavior
Regardless of the subtraction result, SUBLW instruction does not modify the N bit of STATUS register on PIC18F4620. Both PICSimLab versions 0.8.6 and 0.8.7 exhibit the same behavior.

## Steps to Reproduce
1. Download the attached [test.zip](https://github.com/lcgamboa/picsimlab/files/6427908/test.zip) file, and extract it.
2. Create a PIC18F4620 project in MPLABX IDE, select PICSimLab as tool.
3. Add the test.asm file to project.
2. Using the debugger of MPLABX IDE, place a breakpoint at line 73. Start debugging the program.
3. When execution reaches the breakpoint, check the contents of STATUS register, or alternatively advance the execution by one step to observe that it doesn't run as expected.

I'm using MPLAB X IDE v5.40 on Ubuntu 20.04.

lcgamboa, 2021-05-06:
@necrashter, thanks for reporting the problem. 

The error is in the PICSim project (used by PICSimLab) and has already been corrected: https://github.com/lcgamboa/picsim/commit/0a07fd3f5749712c60824d55ef3675b503846cb9

Apparently a coding error put a break statement on the wrong line.

Thank you for helping to improve PICSimLab.

necrashter, 2021-05-06:
Thanks a lot for the quick fix. I compiled and tested the latest version, it now works correctly.

I should have created this issue in PICSim repository. Sorry for the inconvenience.

closed 2021-05-06 by necrashter