lcgamboa/picsimlab #36: Memory leak on v0.8.7 on Windows 10

aabounegm, 2021-05-08:
I downloaded and installed the new update (v0.8.7) today on my Windows 10 (20H2 - Build 19042.928), but its memory consumption is crazy. It keeps allocating about 5 MBs every second, I stopped it after it had already allocated more than 6 GBs of RAM! It barely even reached 10 MBs with v0.8.6
However, this doesn't seem to happen with all boards. I selected Arduino Uno, and the leak stopped (but memory usage didn't go down again). I go back to PICGenios (which is what I need to use) and it goes back to eat more RAM than Google Chrome ever did.
I tried uninstalling and reinstalling from scratch after also removing everything in `%APPDATA%\picsimlab` (I had previously installed it without first removing the previous version, 0.8.6), but that didn't help.
I got the installer from the releases section in this repo (in particular, the [picsimlab_0.8.7_win64_setup.exe](https://github.com/lcgamboa/picsimlab/releases/download/v0.8.7/picsimlab_0.8.7_win64_setup.exe) file).
Please help with this memory leak as I really liked the new SVG pictures, but had to revert back to v0.8.6 just to be able to use it

lcgamboa, 2021-05-08:
Sorry, there was a problem with the lxrad library used by PICSimLab. The problem has now been resolved https://github.com/lcgamboa/lxrad/commit/f5daba44a48f43b1e89f800e1ea2584e3bd2cd9b.

I'm working to update the binaries as soon as possible. I'll let you know here when the update occurs.

aabounegm, 2021-05-08:
Thank you for the extremely quick fix and response! Waiting for the new binaries :grin:

lcgamboa, 2021-05-09:
@aabounegm  please try the new release [picsimlab_0.8.8_win64_setup.exe](https://github.com/lcgamboa/picsimlab/releases/download/v0.8.8/picsimlab_0.8.8_win64_setup.exe) . 

Any problem please contact. I do extensive tests on Linux where I develop, there is not much time left to do tests on Windows.

aabounegm, 2021-05-09:
The memory leak seems to be solved, but the CPU usage looks a bit high. With the clock on my PICGenios (PIC18F4620) set at 1 MHz (the lowest), and even with the board turned off (using the button on the bottom right), it still uses up at least 10%-15% of my (Intel Core i7-8550U) CPU, with the task manager reporting "High" to "Very high" power usage (alternating).
Once I switch it to 40 MHz, if it does not crash, then it stays at more than 30%-35% CPU usage with a constant "Very high" power usage. However, the behavior in the previous versions was similar to this, so I assume it is expected, or at least outside the scope of this issue.
Thank you for fixing the leak. Now it doesn't pass about 15 MB

closed 2021-05-09 by aabounegm

lcgamboa, 2021-05-09:
Currently the PICSimLab redraws everything every 100ms, I am studying only redraw when necessary to reduce the use of future processing. 
Thank you for your considerations.

lcgamboa, 2021-05-17:
The latest binaries of version 0.8.8 have been updated with the modification of only redrawing what is necessary, with that the CPU usage has been reduced when using a low clock frequency.

aabounegm, 2021-05-18:
I noticed that the CPU usage is related to the clock frequency, but it was the same situation even with v0.8.6 (the earliest I've used). I downloaded the updated binaries now, but I see no difference. As I described earlier:
> Once I switch it to 40 MHz, it stays at more than 30%-35% CPU usage with a constant "Very high" power usage

Frankly, I found it strange when you said that "PICSimLab redraws everything every 100ms", even though in my experience the CPU usage was tied to the clock frequency

lcgamboa, 2021-05-18:
PICSimLab uses two threads, one to redraw the screen every 100ms and the other to simulate the microcontroller. In the previous version, the screen was redraw every 100ms, so even with the microcontroller at low frequency or stopped (board off) there was an unnecessary CPU use to redraw the screen. Now only the modification is redrawn, reducing the CPU usage of the draw thread. The microcontroller simulation thread continues to use CPU according to the chosen clock frequency, the higher, the more use as you already said.
The difference is only when the clock is too low or the board is off, for the clock at high values, no significant changes are noticed.

aabounegm, 2021-05-18:
Thanks for the great explanation and help!