lcgamboa/picsimlab #39: Change install location

deananorth, 2021-08-03:
Is it possible to install PICSimLab on a drive other than C: using the installer for Windows?
The installer seems to provide no way of setting the install location.

lcgamboa, 2021-08-03:
Please test this new installer version with the option to choose the install location:  https://sourceforge.net/projects/picsim/files/latest%20code%20build%20%28unstable%29/picsimlab_0.8.9_210803_win64_setup.exe/download

deananorth, 2021-08-04:
Tested the new installer.

It seems to work as expected and installed PicSimLab where I asked. (D:\Program Files\PicsimLab)

[image]

closed 2021-08-04 by deananorth