lcgamboa/picsimlab #40: Compilation Error!!

sammycage, 2021-08-04:
picsimlab4.cc: In member function ‘void CPWindow4::SetSample()’:
picsimlab4.cc:340:4: error: reference to ‘data’ is ambiguous
  340 |    data[fp][0][is] = -pins[0]+((1.0 * rand () / RAND_MAX) - 0.5)*0.1;
      |    ^~~~
In file included from /usr/local/include/c++/11.1.0/string:54,
                 from /usr/local/include/c++/11.1.0/bits/locale_classes.h:40,
                 from /usr/local/include/c++/11.1.0/bits/ios_base.h:41,
                 from /usr/local/include/c++/11.1.0/ios:42,
                 from /usr/local/include/c++/11.1.0/ostream:38,
                 from /usr/local/include/c++/11.1.0/iostream:39,
                 from /usr/include/lxrad/clxrad.h:52,
                 from /usr/include/lxrad/lxrad.h:45,
                 from picsimlab4.h:29,
                 from picsimlab4.cc:29:
/usr/local/include/c++/11.1.0/bits/range_access.h:319:5: note: candidates are: ‘template<class _Tp> constexpr const _Tp* std::data(std::initializer_list<_Tp>)’
  319 |     data(initializer_list<_Tp> __il) noexcept
      |     ^~~~
/usr/local/include/c++/11.1.0/bits/range_access.h:310:5: note:                 ‘template<class _Tp, long unsigned int _Nm> constexpr _Tp* std::data(_Tp (&)[_Nm])’
  310 |     data(_Tp (&__array)[_Nm]) noexcept
      |     ^~~~
/usr/local/include/c++/11.1.0/bits/range_access.h:300:5: note:                 ‘template<class _Container> constexpr decltype (__cont.data()) std::data(const _Container&)’
  300 |     data(const _Container& __cont) noexcept(noexcept(__cont.data()))
      |     ^~~~
/usr/local/include/c++/11.1.0/bits/range_access.h:290:5: note:                 ‘template<class _Container> constexpr decltype (__cont.data()) std::data(_Container&)’
  290 |     data(_Container& __cont) noexcept(noexcept(__cont.data()))
      |     ^~~~

sammycage, 2021-08-04:
But it compiled successfully after I changed `board` to g_board``

lcgamboa, 2021-08-19:
Variable renamed in this commit https://github.com/lcgamboa/picsimlab/commit/cef3c146c96d467ff7967a3db62847c1567d0b9d

closed 2021-08-19 by lcgamboa