lcgamboa/picsimlab #41: Fail compile with g++ 9.3.0

abriotde, 2021-08-07:
Hello, 

I am on Ubuntu 20.04.2 LTS and I have g++ 9.3.0 and I get the errors at Linking:
`
/usr/bin/ld : /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o : dans la fonction « _start » :
(.text+0x24) : référence indéfinie vers « main »
/usr/bin/ld : /tmp/picsimlab.dMLM2C.ltrans24.ltrans.o : dans la fonction « avr_callback_run_gdb_(avr_t*) » :
/home/alberic/bin/picsimlab/src/boards/bsim_simavr.cc:418 : référence indéfinie vers « gdb_watch_find »
/usr/bin/ld : /home/alberic/bin/picsimlab/src/boards/bsim_simavr.cc:425 : référence indéfinie vers « gdb_send_quick_status »
/usr/bin/ld : /tmp/picsimlab.dMLM2C.ltrans24.ltrans.o : dans la fonction « bsim_simavr::DebugLoop() » :
/home/alberic/bin/picsimlab/src/boards/bsim_simavr.cc:505 : référence indéfinie vers « gdb_network_handler »
/usr/bin/ld : /tmp/picsimlab.dMLM2C.ltrans24.ltrans.o : dans la fonction « avr_callback_run_gdb_(avr_t*) » :
/home/alberic/bin/picsimlab/src/boards/bsim_simavr.cc:434 : référence indéfinie vers « gdb_network_handler »
collect2: error: ld returned 1 exit status
`
I tried using clang++ but I have the same issue.

lcgamboa, 2021-08-07:
The problem is not the gcc version or clang++. 
Currently PICSimLab is only compatible with this modified version of simavr:  https://github.com/lcgamboa/simavr

abriotde, 2021-08-07:
Merci c'est ce qu'il me manquait, il tourne ;)