lcgamboa/picsimlab #42: Pwm not showing

soeidat, 2021-08-12:
I am trying to run BWM at 5khz, but it seems that picsimlab cannot generate the same code listed in real ardunio uno where it works perfectly.

lcgamboa, 2021-08-12:
Hi,
PICSimLab uses simavr to simulate arduino uno. Not all PWM modes are implemented in simavr. 
If you can submit a minimal code example of the PWM configuration that works in real Uno and not in the simulator I can try to find the problem and fix it.

soeidat, 2021-08-12:
Hi Icgamboa,

Please find the below code 

#include <PWM.h>
int pin_PWM = 9;
int32_t frequency = 5000; //frequency (in Hz)
void setup() {
  //initialize all timers except for 0, to save time keeping functions
  InitTimersSafe();
  //sets  frequency for the specified pin
  bool success = SetPinFrequencySafe(pin_PWM, frequency);
  if (success) {
    pinMode(13, OUTPUT);
    digitalWrite(13, HIGH);
    pinMode(pin_PWM, OUTPUT);
  }
}
void loop() {
    analogWrite(pin_PWM, 800); //duty cycle 50%
}

lcgamboa, 2021-08-12:
Hi Soeidat,

thank you for the code example.
What PWM library are you using in Arduino IDE? 
There are several PWM libraries.

soeidat, 2021-08-13:
Hi Lcgamboa, 

I am using pwm-master library.

lcgamboa, 2021-08-13:
Using the lib https://github.com/terryjmyers/PWM  the PWM mode 8 is selected. 
This mode is not yet supported by atmega328 on simavr. 
The simplified program below reproduces the problem. 
```
  //Arduino Pin 9 -> PB1
  DDRB |= 0x02;  //PB1 output

  TCCR1A = 0x80; //Clear OC1A on Compare Match, set OC1A at BOTTOM (non-inverting mode)
  TCCR1B = 0x11; // no prescaling , PWM Phase and Frequency Correct (mode 8)
  ICR1 =  0x0640; //Top
  OCR1A = 0x0320; //Duty
```
Changing to PWM mode 10 the program works.  (TCCR1A = 0x82).
I'll take a look at simavr's PWM generation code to try to solve the problem.
When I have more information I'll let you know here.

lcgamboa, 2021-08-15:
Hi @soeidat ,

Apparently the PWM issue in simavr has been fixed with this modification https://github.com/lcgamboa/simavr/commit/f17571687cfa8813d11c51689439115db011c82d.

You can download this version with the fixed simavr for testing [picsimlab_0.8.9_210814_win64_setup.exe](https://sourceforge.net/projects/picsim/files/latest%20code%20build%20%28unstable%29/picsimlab_0.8.9_210814_win64_setup.exe/download)

Please report your test results. 

Thanks

soeidat, 2021-08-17:
Hi Lcgamboa, 

Thanks for your help. It is working now but the PWM signal not stable when it displayed on the Oscilloscope.

lcgamboa, 2021-08-17:
Hi @soeidat ,

writing to the pin direction register always changes the pin value in simavr. 
Since the analogWrite function always sets the pin as output, calling it continuously causes the PWM to malfunction.
It is a limitation of simavr currently. 
Using analogWrite once in setup or adding a delay in the loop solves the problem in the simulation. 
```
void loop() {
  analogWrite(pin_PWM, 800); //duty cycle 50%
  delay(10);//fix the PWM malfuction in simulation
}
```
Without delay():
[image]

Witht delay():
[image]