lcgamboa/picsimlab #44: picsimlab awfully slow (arduino local version)

atesin, 2021-09-06:
hi... i am trying to run a virtual arduino with this configuration for testing...
```
  terminal           picsimlab     uart             terminal
                     .-------,
  ,======,  com0com  |]  #   |             com0com  ,======,
  |>_    |           |       ',    ,----,           |>_    |
  |      |-----------|]   ##  |----|  # |-----------|      |
  '======'           '--------'    '----'           '======'
```
i try to set like real parameters (16mhz)...  but it runs at 0.02% speed!! ... and eats 20% of my cpu time!
i don't know why is so slow, this is just an arduino... my playstation2 emulator runs faster with all enabled
i have a pentium xeon 3.6ghz 12 cores, 12gb ram, windows 8.1 x64

... is there any optimization i could do??

lcgamboa, 2021-09-06:
Hi @atesin,

it's probably some problem with the IO uart part. 
I still haven't had time to optimize all the parts after I implemented the simulation speed adjustment. 
If you can submit a minimal sketch that reproduces the problem I can try to solve this issue.

atesin, 2021-09-06:
i noticed (but not sure) when i put virt arduino at 16mhz its stats ok, about 1.0x - 0.9x emulation speed
when i OPEN the virt uart mod (without even attaching it yet) the performance drops to 0.1x and continues dropping....

then even if i close everything but the arduino the performance still drops and cant recover it, i had to finally close everything losing some work

i am trying to write a python esp8266 serial simulator since the included one is very limited though, and i need this environment to being testing it

anything i can help (testing, etc) i am here

lcgamboa, 2021-09-06:
Hi @atesin , 

I have modified the reading of the data available status on the serial port so that it doesn't happen every simulation cycle. With that the problem seems to have been removed.  7ff71ece54eab52749516d84777e8324bbcb0546

Please check using the test version  https://sourceforge.net/projects/picsim/files/latest%20code%20build%20%28unstable%29/picsimlab_0.8.9_210906_win64_setup.exe/download

Indeed the esp8266 modem simulation is limited and apparently no one normally uses it. As I don't have much free time to improve PICSimLab, I'm giving preference to solving problems reported by users.

Thanks for reporting the issue and helping to improve the simulator.

atesin, 2021-09-06:
hi...  thanks for your time .... i downloaded, reinstalled and tried this test version .... but unfortunately until i think i have all properly configurated, it doesn't seem to work (terminal = 'failed to initialize the port').....  unless i did something wrong

- arduino rx, tx = pins 10, 11 4800 baud (softwareserial is 8N1 by default)
- uart rx, tx = pins 11, 10
- uart com45 4800 baud (*)
- com0com = com45-com44
- terminal = com44 4800 baud, "failed to initialize the port"

(*) in uart virt mod, you could also remove the need to SET THE BAUD RATE (unless needed for other boards), letting the endpoints configure its baud rates, turnning the uart mod acting just like a bridge like com0com does

[image]

lcgamboa, 2021-09-06:
Hi @atesin ,

by your screenshot the PICSimLab connected to port 32 and the IO UART at port 45 without problems. 
The serial port settings are correct as well.
The problem may be the termite, several serial terminal programs for Windows only operate serial floods from COM1 to COM9. Because of restrictions on the Windows API:  https://support.microsoft.com/en-us/topic/howto-specify-serial-ports-larger-than-com9-db9078a5-b7b6-bf00-240f-f749ebfd913e

The virtual UART uses Baudrate to set the real serial port speed and read the simulation serial output pins. 

For COM0COM it is not necessary to set the Baudrate correctly if the baudrate emulation is not activated.

atesin, 2021-09-07:
hello again
- if worked be before with this same setup, but freezing slow
- maybe that windows restriction is an old limitation (doesnt point the date or systems affected), since you will see below com32-com33 are working
- the com32 you see at lower corner of arduino is HardwareSerial (arduino usb-serial port), not shown but it works ok
- is baud conversion strictly needed?, i see com0com works ok just like a pipe, maybe conversion is not really needed since you anyway have to configure serial signal at endpoints... what would happen if virt uart will just also 'copy' the analog signals directly between endpoints?, will it work?, at least making it optional?

in this image you can see the whole environment... you can see arduino hardwareserial (usb) -> com33 -> com0com -> com32 -> arduino ide serial monitor, all with common config is working fine ... the other line however dont

[image]

i am glad to could help

lcgamboa, 2021-09-07:
Hello again, 

I have tested this example https://lcgamboa.github.io/picsimlab_examples/board_Arduino_Uno.html#board_Arduino_Uno_atmega328p_softserial on windows with termit (with com44 and com45 )and it work without problem.

Make sure you have not left any other programs connected to the com44 port, there can only be one program connected at a time on each serial. 

Yes, the baudrate is necessary as the IO uart reads the signal generated by the serial software directly from the pins, without the timing it is not possible to read the serial signal correctly. 

I have made the same test below on windows:

[image]

atesin, 2021-09-07:
i feel so stupid, i had the port already opened.... take advantage to get a glimpse of my project... is just a dumb simulator that gets some text and answers another according

now that i closed all other serial ports, i am using this new picsimlab version, and when i "connect" to arduino pins, it hangs (mouse pointer with a wheel)

[image]
[image]

( my esp8266 text simulator v0.000001alphaalphaalpha, that was blocking the ports com44-com45 )

[image]

lcgamboa, 2021-09-07:
PICSimLab sometimes freezes if the other com0com port has not been open previously. 
This problem only occurs sporadically on WIndows and I still haven't been able to determine the origin (whether it is in PICSimLab or com0com). 
Opening the com44 port before PICSimLab connects to com45 should make it work. 

Will your esp8266 text simulator only simulate AT commands or will it emulate the wifi too?

atesin, 2021-09-07:
i open all the consoles before, then open picsimlab, then the virt uart, then configure it... it don't hangs, but i resetted the virt arduino many times, and the arduino ide serial monitor displays the message but the cutecom side don't display anything, also the "chat" function doesnt work, as if the signal were always low.... idk what is happening, i hope i didnt commit some other mistake again :P , i checked all many times 

[image]

btw, my esp simulator project currently uses the pySerial lib (i am learning python)...  and later i would like to give the ability to open tcp sockets through your existing pc connection as netcat does... but many functions will have to be simulated or unsupported as wifi, energy or fota for example

i would like a tool like your picsimlab that emulates the whole esp8266 cpu that we can even flash it =D ... or a decent api-level simulator at least, but i haven't found any :.(

lcgamboa, 2021-09-07:
Hi @atesin 

from the screenshot everything seems to be set up correctly. If you prefer, you can enter discord and stream your computer screen so we can try to find out where the problem is.  https://discord.gg/fMT8szFYq7

The esp8266 (and the esp32) are very complex SOCs and relatively inexpensive, as not all hardware is documented it would not be simple to make a complete simulator like the PIC and AVR simulators used by PICSimLab.
 Expressif is currently migrating from esp8266 to ESP32-C3 risc-V, so the latter should replace esp8266.
 I only know qemu-stm32 capable of simulating an ESP32, but the wifi part is not supported yet https://github.com/espressif/qemu/wiki

closed 2021-09-10 by lcgamboa

atesin, 2021-09-12:
eeh....  i installed this alpha version in my notebook, and the problem still persists ... maybe new installations suffer of this ... i will reinstall and try again