lcgamboa/picsimlab #45: [bug] picsimlab freeze when closing

atesin, 2021-10-05:
when i click the "X" at title bar it freezes and mouse pointer turns a wheel... after some more attempts it shows this dialog

[image]

environment:
- windows 8.1 pro x64 spanish (i hate w10)
- picsimlab v0.8.9_210925
- intel xeon 3.5ghz, 12gb ram, nvidia gtx1650 4gb

lcgamboa, 2021-10-05:
Can you repeat the problem multiple times? 
Could you submit the .pzw file and code or a minimal example that causes the problem? 
I need to reproduce the problem in order to detect the cause.

lcgamboa, 2021-10-07:
this problem is recurrent and linked to the use of com0com. It only happens sporadically to me (I don't use windows normally) and I can't reproduce the problem to try to detect the error. 
If you can save the workspace and upload the .pzw file it would be a start to try to reproduce the problem.

mubeen-jukaku, 2021-11-23:
This is reproducible on Windows 10.
I tried using this example https://lcgamboa.github.io/picsimlab_examples/pzw/board_McLab2/PIC16F777/test_b3.pzw
Kindly check.

lcgamboa, 2021-11-23:
Hi @mubeen-jukaku, I tested this example here with the latest stable (0.8.9) and development version of PICSimLab in Windows 10 and I couldn't reproduce the error. There may be some different configuration in Windows that causes the problem on some computers and not others. Are you using the buffer overrun option in com0com as shown in the [documentation](https://lcgamboa.github.io/picsimlab_docs/stable/Com0comInstallationandConfigurationWindows.html#x36-350006.1)?

mubeen-jukaku, 2021-11-24:
Thanks for your suggestion. I tried to reinstall and reconfigure com0com driver. The ports are not getting detected. In device manager it shows an error saying the driver is not signed. Is there a solution for this. I am searching for free alternatives as well.

atesin, 2021-11-24:
in my case, i remember i disabled buffer overrrun protection, because com0com always transfers data ok no matter the baud rates i set in each endpoint, and sometines in my app i need to see the "garbage" to know when bauds+flags are wrong.... and i thought this setting could be related

(i don't have my pc right now because am in another city temporarily)

lcgamboa, 2021-11-24:
> Thanks for your suggestion. I tried to reinstall and reconfigure com0com driver. The ports are not getting detected. In device manager it shows an error saying the driver is not signed. Is there a solution for this. I am searching for free alternatives as well.

When the latest version of com0com doesn't work, the previous version usually resolves the issue of certificates. Try to install it https://sourceforge.net/projects/com0com/files/com0com/2.2.2.0/com0com-2.2.2.0-x64-fre-signed.zip/download

lcgamboa, 2021-11-24:
> in my case, i remember i disabled buffer overrrun protection, because com0com always transfers data ok no matter the baud rates i set in each endpoint, and sometines in my app i need to see the "garbage" to know when bauds+flags are wrong.... and i thought this setting could be related
> 
The "buffer overrun" option is usually useful to discard data that is received without the two ports being connected (common behavior in a real device). With this option turned off when the port is opened it receives all the buffer contents that were written by the other peer when it was closed. 
> (i don't have my pc right now because am in another city temporarily)
Any test or information that helps to identify the source of the problem is welcome.

mubeen-jukaku, 2021-11-24:
> When the latest version of com0com doesn't work, the previous version usually resolves the issue of certificates. Try to install it https://sourceforge.net/projects/com0com/files/com0com/2.2.2.0/com0com-2.2.2.0-x64-fre-signed.zip/download

I was able to install the older version. But picsimlab is picking up COM5 port by default. This port is a Serial Over Lan (SOL) port. Is there a way to specify the port to pick in picsimlab?

lcgamboa, 2021-11-24:
>Is there a way to specify the port to pick in picsimlab?

Yes. In menu File->Configure.

sulemankm, 2023-11-11:
Hi, It still freezes for me on windows 10, when I change a board from the menu. 

NOTE: I'm also using com0com.

lcgamboa, 2023-11-11:
Hi @sulemankm , 
this problem normally occurs when com0com's buffer overrrun option is not enabled. 
Could you tell us which exact version you are using (0.9.0?) and which board change causes the problem to happen? 
I need as much information as possible to try to reproduce the problem. 
After crashing when opening PICSimLab, does any information appear in the Log window?

closed 2023-12-07 by lcgamboa

Ranjaniot, 2024-01-28:
# lcgamboa
[image]
[image]
I completed the project 
[Ranjaniot](https://github.com/Ranjaniot)
/
[IOT-based-Home-Automation-Solution-in-PicsimLab](https://github.com/Ranjaniot/IOT-based-Home-Automation-Solution-in-PicsimLab)
[image]