lcgamboa/picsimlab #47: Ola, eu instalo e desinstalo o picsim para win 10 mas no Breadbord, quando acionado o Spare parts a tela fica toda cinza, nao aparece nada

Ricarnet66, 2021-11-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files**
If applicable, add a minimal source code together with a PICSimLab workspace (.pzw file) to help reproduce your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Troubleshooting:**
The simulation in PICSimLab consists of 3 parts:

- The microcontroller program
- Microcontroller simulation (made by [picsim](https://github.com/lcgamboa/picsim) and [simavr](https://github.com/buserror/simavr))
- Simulation of boards and parts

When a problem occurs it is important to detect where it is occurring.

One of the most common problems is the error in the microcontroller program. Before creating an issue, test your code on a real circuit (even partially) to make sure the problem is not there.

Errors in the microcontroller simulation can be detected using code debugging. Any instruction execution or peripheral behavior outside the expected should be reported in the project of simulator used ([picsim](https://github.com/lcgamboa/picsim) or [simavr](https://github.com/buserror/simavr)).

If the problem is not in either of the previous two options, the problem is probably in PICSimLab. A good practice is to send a source code together with a PICSimLab workspace (.pzw file) to open the issue about the problem.

lcgamboa, 2021-11-02:
Por favor envie mais informações e se possível um screenshot do problema.

lcgamboa, 2021-11-09:
Fechado por falta de informação. Caso seja enviadas informações que ajudem identificar o problema será reaberto.

closed 2021-11-09 by lcgamboa