lcgamboa/picsimlab #48: AppImage version doesn't launch.

artcamir, 2021-11-26:
**Describe the bug**
AppImage version doesn't launch in Fedora 35
I tested it with and without AppImageLauncher.

**To Reproduce**
Steps to reproduce the behavior:
1. Download any AppImage version
2. Give it the permissions to run
3. Run the app
4. See error

**Expected behavior**
The App runs and works as it should do.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Fedora with GNOME]
 - Version [35]

lcgamboa, 2021-11-26:
Apparently the Appimage format version doesn't work directly on newer Linux versions, it is necessary to specify the libgmodule library path.
try:
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgmodule-2.0.so ./PICSimLab-0.8.9-x86_64.AppImage`

I have fixed this issue in continuous build version [PICSimLab-0.8.9_211123-x86_64.AppImage](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab-0.8.9_211123-x86_64.AppImage) check to see if this version works for you.

ArtMiranda, 2021-11-26:
Using the new build version it launches now, but when i click to load a hex file it crashes, and if i try to launch it again it doesn't launch.

If i download the file again and try to launch it works, but it opens just once.

lcgamboa, 2021-11-26:
Is there an error message on the console? Is the error when opening the file loading dialog or loading the file? 

Unfortunately I don't have a Fedora available to test, I do the tests on Debian and Ubuntu.

ArtMiranda, 2021-11-26:
Here's a screenshot of when i run it at the terminal

[image]

lcgamboa, 2021-11-26:
The AppImage format hasn't worked very well on newer versions of Linux... I'll probably use another format like Flatpak in the future. From the screenshots I can't identify what the problem would be. I'll do more research. If you want we can talk (in Portuguese) through the [Discord](https://discord.gg/fMT8szFYq7) channel later and try to solve the problem.

ArtMiranda, 2021-11-26:
I think Flatpak would be the best way on my opinion, How can I contact on Discord?

ArtMiranda, 2021-11-26:
I think Flatpak would be the best format, I'll contact on Discord.

ArtMiranda, 2021-12-03:
It opens now, but there is an error, any changes in the window needs the window to "reload" (Alt+tab twice), as show in the video below.

(This program should do a binary count with the LEDS)

https://user-images.githubusercontent.com/50926874/144537118-963c849d-8d9b-4069-8944-42ec6e288dd4.mp4

Maybe this is an error with the Wayland integration, I'll do some tests on Xorg and report to you.

ArtMiranda, 2021-12-03:
It opens now, but there is an error, any changes in the window needs the window to "reload" (Alt+tab twice), as show in the video below.

(This program should do a binary count with the LEDS)

https://user-images.githubusercontent.com/50926874/144537258-8d92fef0-d749-4305-b486-9ce93db84943.mp4

Maybe this is an error with the Wayland integration, I'll do some tests on Xorg and report to you.

ArtMiranda, 2021-12-03:
Tested on Xorg, works perfectly.

The bugs are problems with the Wayland integration.

lcgamboa, 2021-12-03:
The problem is in PICSimLab, apparently it only works in wayland by setting the environment variable GDK_BACKEND=x11. But I haven't been able to get this to work with flatpak yet. 
When I compile natively in ubuntu it works without problem in wayland. It's some configuration in one of the wxwidgets, gtk3 or gdk libraries.

lcgamboa, 2021-12-03:
Can you try this command: 
`flatpak run --nosocket=wayland com.github.lcgamboa.picsimlab`

ArtMiranda, 2021-12-03:
Now it works perfectly in wayland.

I edited my PicSimLab shortcut with MenuLibre and it works as it should now.

Do you have plans to upload this flatpak version on flathub?

https://user-images.githubusercontent.com/50926874/144677933-e2be3f45-a8a1-42f5-9e39-37692de54009.mp4

closed 2021-12-03 by ArtMiranda

lcgamboa, 2021-12-04:
Yes, there is an open issue https://github.com/lcgamboa/picsimlab/issues/49 with some fixes to make before submitting to flathub. 
I updated flatpak and disabled wayland, now it works without having to pass parameters.

Ranjaniot, 2024-01-20:
> Apparently the Appimage format version doesn't work directly on newer Linux versions, it is necessary to specify the libgmodule library path. try: [...]

[image]
[image]
helpful the embedded system design