lcgamboa/picsimlab #49: simavr tagging and naming scheme

tinywrkb, 2021-11-27:
In the process of creating an initial Flatpak packaging for picsimlab I'm also adding proper entries in `release-monitoring.org` (Anitya) as it helps update automation of the Flatpak manifest, and it's also a useful tool for distro maintainers.

I already added entries for [picsimlab](https://release-monitoring.org/project/241290/), [lxrad](https://release-monitoring.org/project/241292/), and [lunasvg](https://release-monitoring.org/project/241295/), but I haven't added yet a `picsimlab-simavr` entry as there's no tag naming scheme that could help differentiate upstream `simavr` releases from the `picsimlab-simavr` releases.

There's a `1.6a` `picsimlab-simavr` release, which is a bit confusing way to tag a fork release.  
I suggest to tag `picsimlab-simavr` releases as `picsimlab-MAJOR.MINOR.PATCH` as Anitya should be able to filter tags with a prefix. 
 
I'm not sure about tagging releases as `picsimlab-vMAJOR.MINOR.PATCH`. I guess it should also work, but I recommend against it, as I haven't confirmed that the Anitya filter does not split on dash/hyphen, and Anitya has its quirks.

p.s. I suggest turning on the issues in the simavr fork, as long as it's expected to be shipped with picsimlab, and not just a downstream development fork.

lcgamboa, 2021-11-27:
I am currently trying to create a flatpak for PICSimLab, but I have no experience. I don't know if it's possible to use lunsavg static library without installing in flatpack which is needed to build lxrad library...
 As for picsimlab-simavr I will fix the tag. In the future I intend to use the original simavr, I've already started to make some changes for that, but currently I'm out of time.

tinywrkb, 2021-11-27:
I actually might have something working a bit later today.

lcgamboa, 2021-11-27:
I will wait.
I've already updated the picsimlab-simavr tags.
I should also use flatpak for continuous build releases, I'm using your flatpak repository as a source of examples to study.

tinywrkb, 2021-11-28:
Here's a [very initial package](https://github.com/tinywrkb/flatpaks/blob/master/io.github.lcgamboa.picsimlab/com.github.lcgamboa.picsimlab.yaml), just get it to be built and packaged.  
It took a bit more effort than the usual packaging, so it's not pretty yet.

I think that bundling CuteCom might not be the right approach, so I added a flatpak-spawn wrapper instead. I haven't tested it yet, so I'm not sure that it works.  
Using a flatpak-spawn wrapper means enabling the very permissive `--talk-name=org.freedesktop.Flatpak` permission, which isn't a great choice, and shouldn't be enabled by default. Instead, let the user set it manually with an override. 
Maybe add support for [GTKTerm](https://github.com/Jeija/gtkterm/), it would be a better choice for a bundled terminal emulator as it uses GTK3, so no need to switch to the KDE runtime.  
If we bundled a terminal emulator, then I think we should avoid exporting its desktop file, so the user won't be confused by having two entries of the same terminal emulator pop up in the desktop environment launcher.  
I already have GTKTerm packaged, so it would be pretty easy to bundle it.

tinywrkb, 2021-11-28:
And forget to mentioned that I plan to package CuteCom later, and suggest to upstream to publish on Flathub.

lcgamboa, 2021-11-28:
I use cutecom because I compiled it for windows, I tried it first with gtkterm but I couldn't cross compile.
I can change PICSimLab to support configuring which serial terminal to use for the next version. 
I think you can change the cutecom command to gtkterm in the picsimlab1.cc file using sed to make it easier to implement until the next version comes out. 
I just made my flatpak for now too (yours is much more refined), but I used git instead archive in source type to build with the latest versions. 
It will be necessary to disable several things in wxgtk that are not used by the lxrad library and by PICSimLab. 
In your file you need to update the simavr version from v1.6a to v1.7 to work (and use install-simavr instead install to remove freeglut and examples). The tag v1.6a as been removed .
I have never used a flatpak before! I started by creating one.

tinywrkb, 2021-11-28:
> I think you can change the cutecom command to gtkterm in the picsimlab1.cc file using sed to make it easier to implement until the next version comes out.

Right, I noticed this in the source. I will switch then to GTKTerm and bundle it in the package.

> I just made my flatpak for now too

That's amazing!

> I have never used a flatpak before! I started by creating one.

 Flatpak is a pretty powerful tool to have and master.  
As a user, it brings me control on the apps installed in my system.  
Having apps coupled to the distro is a major design flaw, and Flatpak allows me to decouple them, easily roll back versions or mask updates.  
As a developer, you can ship your apps via distro agnostic platform, knowing the user will run it with the same set of shared libs versions / runtime used for development and testing.

> but I used git instead archive in source type to build with the latest versions.

I prefer using archive sources as flatpak-builder always check remotes when using git source types, and if the remote is not reachable, then the build process will break.  
When using archive source type, flatpak-builder will first look for a checksum match in the local cache.

For local development, dir source type might be a better choice than git in some cases, but [currently it's not being checksummed](https://github.com/flatpak/flatpak-builder/issues/356), so it rebuilds modules with dir sources on every flatpak-builder run.  

You can also manually run the `flatpak build` commands, but this is more involved, and I played with little with this, so I'm not in position to recommend it.

> It will be necessary to disable several things in wxgtk that are not used by the lxrad library and by PICSimLab.

Right, my wxgtk modules is generic, as I use it with other packages. When packaging for Flathub, the package should be minimal.

> and use install-simavr instead install to remove freeglut and examples

Do we need `avr-gcc`? I started packaging it as a module in a Flatpak app, but it's not really the right approach, as it should be packaged as runtime SDK extension, which will also avoid having to build it when building the Flatpak app.  
I think I might create later a `org.freedesktop.Sdk.Extension.avr` packaging, and if I see that there might be possible users (apps) for it, then I will send a PR to Flathub.

lcgamboa, 2021-11-28:
> Do we need avr-gcc?

For PICSimLab work and package no. Change the picsimlab-simavr  form "make install" to "make install-simavr" remove this dependency.

tinywrkb, 2021-11-28:
Thanks.  

I added GTKTerm to the package, maybe it should be moved up in the dependency tree to avoid rebuilding it when picsimlab is updated, as it should be better for local development.

I'm not sure how well the app is working yet, but except for testing the app, I think the other issues that should be resolved before Flathub publishing:

* [ ] Remove workarounds needed to build the package, and fix them in the sources' buildsystem
  * [ ] Use reverse DNS naming scheme for dekstop files and icons filenames
  * [ ] Drop chmod from the install target, use the install command with mode instead of mkdir and cp call
  * [ ] Don't update mime and desktop DBs. This should be handled by the distro package manager when installing packages to the local machine.
  * [ ] Don't strip binaries. This should be handled by the distro builder/packaging system. As a developer, you should want to make it easy for users to give you coredumps with debugging symbols, and Flatpak makes it easy by automatically create debug extension for apps, so the user just need to install the debug extension.
  * [ ] Install the mime types config. I haven't looked into this at all, it might need a file name update (same as desktop file) to be exported by Flatpak, and maybe other changes (name entries, ...).
* [ ] Add AppStream appdata/metainfo
* [ ] Consider linking against shared lunasvg lib. This is more in line with distro packaging, and be future proofed against possible licensing changes.
* [ ] Minimal wxGTK lib.
* [ ] [f-e-d-c](https://github.com/flathub/flatpak-external-data-checker/) properties for all the modules to help automating updates.
* [ ] Switching tagged releases.
* [ ] Cleanup. Drop everything that is not needed for the desktop app.

lcgamboa, 2021-11-29:
> Don't update mime and desktop DBs. This should be handled by the distro package manager when installing packages to the local machine.

PICSimLab has an install option "install_app" which does not update the mime and desktop DBs. 
It is necessary to change or you can use "install_app" instead of "install" ?

https://github.com/lcgamboa/picsimlab/blob/5a2dd4a832d269cf3245585f9cab715f93de40e2/src/Makefile#L59

From what I understand of flatpak it is not possible to run any application installed on the system or on another flatpak. All executables must be inside the same flatpak, correct?
In this case it would be necessary to also include gedit, gtkwave (or pusleview) ...

tinywrkb, 2021-11-29:
> It is necessary to change or you can use "install_app" instead of "install" ?

I can use the `install_app` target.

> From what I understand of flatpak it is not possible to run any application installed on the system or on another flatpak. All executables must be inside the same flatpak, correct?

It's possible to use a flatpak-spawn wrapper, like the ]CuteCom wrapper](https://github.com/tinywrkb/flatpaks/commit/6f00e58a298a17c0268e98f890809e2280298036) that I added before bundling GTKTerm, but the permission to run host apps is pretty permissive.  

> In this case it would be necessary to also include gedit, gtkwave (or pusleview) ...

If there's mimetype association, then shouldn't `xdg-open` be enough?  So maybe avoid launching gedit directly and let `xdg-open` do its job.  

Note that the `xdg-open` in the Flatpak sandbox is actually using the [AppChooser Portal](https://flatpak.github.io/xdg-desktop-portal/portal-docs.html#gdbus-org.freedesktop.impl.portal.AppChooser), and AFAIK you can actually have for every Flatpak app a different setting of a default application for specific mimetype.  Flatpak App A would have gedit as the default text editor, while Flatpak app B could have gvim as the default text editor.  
When using `xdg-run, the AppChooser portal will ask the user to select the application to open the file with, and IIRC, after 3 consecutive selections of the same app, it will be set as the default and the user won't be asked again.  
I'm not sure what control the user have on the AppChooser Portal settings in a desktop environment, as I'm not running one (just Sway WM), but you can interact with it directly using the `flatpak permission-` commands.  
Calling this command `permission` is quite confusing, but it changes the AppChooser settings. 

I wouldn't bundle PulseView, libsigrok brings with it quite a bit of dependencies.  
If the app can launch GTKWave, then yes, it should be bundled, at least as long as it's not on Flathub.

lcgamboa, 2021-11-29:
Yes, I can put a compilation option to use xdg-open to open the .txt and .vcd files instead of calling the programs directly. I already do this to open the help urls. 
Thanks for the info.

lcgamboa, 2021-11-30:
Would using a fake ".sterm" extension associated with cutecom or gtkterm (which are not associated to open any file type) in order to use xdg-open be a very bad solution? By my tests it works without problems.

lcgamboa, 2021-11-30:
The above solution was implemented in the commit https://github.com/lcgamboa/picsimlab/commit/22d5302630b15a2f8333bedca6a45b6dfd163681

To enable this support use `make  CXXFLAGS="-DUSE_XDG_OPEN"

closed 2023-12-07 by lcgamboa