lcgamboa/picsimlab #50: Problema com o RTC

Rodolfoloc, 2021-11-28:
I modify the RTC date and clock, but it reverts to a default value. Below is the code where I change it to 12/2/20, 10:40:15, but what appears on the LCD is 12/31/69, 21:00:01. I believe it's a simulator bug. Thanks for the help!

#define _XTAL_FREQ 8000000
#include "configbits.h"
#include <xc.h>
#include "flexlcd.h"
#include <stdio.h>
#include "i2c.h"
#include "rtc.h"
#include "eeprom.h"
#include "eeprom_ext.h"

//Variables
unsigned int duty = 100, count = 0; // 0 a 255
unsigned char str[5];
// saídas (colunas), ativadas por 0
#define C1 LATBbits.LB0
#define C2 LATBbits.LB1
#define C3 LATBbits.LB2

// entradas (linhas), com pull-up externo
#define L1 PORTDbits.RD3
#define L2 PORTDbits.RD2
#define L3 PORTDbits.RD1
#define L4 PORTDbits.RD0

void main(void) 
{
    //ADCON0 = 0b00000000; // Configura todos pinos para digital
    ADCON1 = 0b00001111; // desabilita o conversor A/D
    //ADCON2 = 0b00000000;    
    //INTCON2 = 0b00000000;
    INTCON2bits.RBPU = 1; // Pull-up resistors off
    T1CON = 0b10000011; // configura Timer1
    //PIE1bits.TMR1IE = 1; // TMR1 enable interrupt
    //PIR1bits.TMR1IF = 0; // TMR1 interrupt flag
    TMR1L = 0; 
    TMR1H = 0; // valor inicial
    
    // Input or Output
    TRISA = 0xFF;
    TRISB = 0b00000000; 
    TRISC = 0b00000001;
    TRISD = 0b00000000; 
    TRISE = 0b00000000;
    // Clear ports
    PORTA = 0; 
    LATA = 0;
    PORTB = 0; 
    LATB = 0;
    PORTC = 0; 
    LATC = 0;
    PORTD = 0; 
    LATD = 0;
    PORTE = 0; 
    LATE = 0;
    
    //adc_init(); // configura registradores ADCON
    Lcd_Init();
    Lcd_Cmd(LCD_CURSOR_OFF); 

    i2c_init();
    i2c_start();
    
    //day, mth, year, dow, hr, min, sec)
    rtc_w(02, 12, 20, 4, 10, 40, 15);
    __delaywdt_ms(200);
    

    e2prom_w(0, 33); // interna
    e2pext_w(0, 47); // externa
    
    while(1) 
    {
        CLRWDT();        
        
        rtc_r(); // ler valores date e time
        __delaywdt_ms(50);
            
        Lcd_Out(1, 0, date);
        Lcd_Out(2, 0, time);

        __delaywdt_ms(2000);
    }
    return;
}

lcgamboa, 2021-11-28:
Can you share the code of the rtc_w function?

Rodolfoloc, 2021-11-28:
rtc.h

/* ########################################################################

   PICsim - PIC simulator http://sourceforge.net/projects/picsim/

   ########################################################################

   Copyright (c) : 2015  Luis Claudio Gambôa Lopes

   This program is free software; you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation; either version 2, or (at your option)
   any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program; if not, write to the Free Software
   Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.

   For e-mail suggestions :  lcgamboa@yahoo.com
   ######################################################################## */

extern volatile unsigned char date[10];  /*dd/mm/aa*/
extern volatile unsigned char time[10];  /*hh:mm:ss*/

void rtc_w(unsigned char day, unsigned char mth, unsigned char year, unsigned char dow, unsigned char hr, unsigned char min, unsigned char sec);

void rtc_r(void);

rtc.c

/* ########################################################################

   PICsim - PIC simulator http://sourceforge.net/projects/picsim/

   ########################################################################

   Copyright (c) : 2015  Luis Claudio Gambôa Lopes

   This program is free software; you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation; either version 2, or (at your option)
   any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program; if not, write to the Free Software
   Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.

   For e-mail suggestions :  lcgamboa@yahoo.com
   ######################################################################## */

#include"rtc.h"
#include"i2c.h"

volatile char date[10];
volatile char time[10];

unsigned char getd(unsigned char nn)
{
 return ((nn & 0xF0)>>4)+0x30;
}

unsigned char getu(unsigned char nn)
{
  return (nn  & 0x0F)+0x30;
}

unsigned char bin2bcd(unsigned char binary_value)
{
  unsigned char temp;
  unsigned char retval;

  temp = binary_value;
  retval = 0;

  while(1)
  {
    // Get the tens digit by doing multiple subtraction
    // of 10 from the binary value.
    if(temp >= 10)
    {
      temp -= 10;
      retval += 0x10;
    }
    else // Get the ones digit by adding the remainder.
    {
      retval += temp;
      break;
    }
  }

  return(retval);
}
//
void rtc_w(unsigned char day, unsigned char mth, unsigned char year, unsigned char dow, unsigned char hr, unsigned char min, unsigned char sec)
{
  sec &= 0x7F;
  hr &= 0x3F;

  i2c_start();
  i2c_wb(0xD0);            // I2C write address
  i2c_wb(0x00);            // Start at REG 0 - Seconds
  i2c_wb(bin2bcd(sec));      // REG 0
  i2c_wb(bin2bcd(min));      // REG 1
  i2c_wb(bin2bcd(hr));      // REG 2
  i2c_wb(bin2bcd(dow));      // REG 3
  i2c_wb(bin2bcd(day));      // REG 4
  i2c_wb(bin2bcd(mth));      // REG 5
  i2c_wb(bin2bcd(year));      // REG 6
  i2c_wb(0x80);            // REG 7 - Disable squarewave output pin
  i2c_stop();
}

//--------------------- Reads time and date information from RTC (PCF8563)
void rtc_r(void) 
{
  unsigned char tmp;

  i2c_start();
  i2c_wb(0xD0);
  i2c_wb(0);

  i2c_start();
  i2c_wb(0xD1);
  tmp= 0x7F & i2c_rb(1); //segundos
  time[5]=':';
  time[6]=getd(tmp);
  time[7]=getu(tmp);
  time[8]=0;

  tmp= 0x7F & i2c_rb(1); //minutos
  time[2]=':';
  time[3]=getd(tmp);
  time[4]=getu(tmp);

  tmp= 0x3F & i2c_rb(1); //horas
  time[0]=getd(tmp);
  time[1]=getu(tmp);

  i2c_rb(1); //dia semana

  tmp= 0x3F & i2c_rb(1); //dia
  date[0]=getd(tmp);
  date[1]=getu(tmp);

  tmp= 0x1F & i2c_rb(1); //mes
  date[2]='/'; 
  date[3]=getd(tmp);
  date[4]=getu(tmp);

  tmp=  i2c_rb(0); //ano
  date[5]='/';
  date[6]=getd(tmp);
  date[7]=getu(tmp);
  date[8]=0;

  i2c_stop();

}

lcgamboa, 2021-11-29:
I no longer had this RTC writing code that I had written myself!
I found only one bug regarding the year value, which was added by 1 while writing. It has been corrected. 

I used the program below to test:
```
#include "config.h"
#include "lcd.h"
#include "itoa.h"
#include "i2c.h"
#include "rtc.h"

void setup() {
    lcd_init();
    lcd_cmd(L_NCR);
    lcd_cmd(L_L1);
    lcd_str("teste RTC");
    delay(1000);
}

void loop() {

    //teste RTC
    lcd_cmd(L_CLR);
    for (int i = 0; i < 20; i++) {
        rtc_r();
        lcd_cmd(L_L1);
        lcd_str((const char *) date);
        lcd_cmd(L_L2);
        lcd_str((const char *) time);
        delay(500);
    }

    lcd_cmd(L_CLR);
    lcd_cmd(L_L1);
    lcd_str("Ajuste RTC");
    delay(1000);
    //day, mth, year, dow, hr, min, sec)
    rtc_w(02, 12, 20, 4, 10, 40, 15);
}

```
  and it worked as shown below:
[image]

If you want the project of code that I had used, get in touch by email or discord.

Rodolfoloc, 2021-11-29:
It didn't work for me. Please, send me the code project you used.

lcgamboa, 2021-11-29:
I need your email address to send the code. Github don't allow send files to others users.

Rodolfoloc, 2021-11-29:
Sorry!

***@***.***

lcgamboa, 2021-11-29:
github doesn't allow you to send email addresses directly either! Below is a dropbox link with the project. Make sure it works for you. I tested it on the latest version of PICSimLab. 

https://www.dropbox.com/sh/jk3xqilwufxpc8f/AABEvC8gcEf8CY4VPtImUQ9ya?dl=0

closed 2022-03-02 by lcgamboa