lcgamboa/picsimlab #52: [question/suggestion] save/load memory state to/from file

atesin, 2021-12-17:
imagine this situation...

suppose i have 10 arduinos, each one with a different sketch loaded, and different eeprom contents (and maybe different modules connected too)..... is there a way to simulate this in picsimlab too? ... i mean "interchange" arduino states (flash+eeprom) through memory  dump files or something?

.... going further... could different picsimlab instances with different "states" run and interact each other?

atesin, 2021-12-17:
hi...  thanks for your early response

so, that mean the option for backup/restore different projects (with firmwares) is "save/load workspace" ??? ...  with flash (eeprom not yet) and all attached devices and cabling too?

cool

....  this is one of my prototypes... having picsimlab really helps me to agilize my development process, by testing single functions and code snipped before flashing the actual sketch in the device

[image]

supposedly it will be a cheap irrigation system for little farmers.... it has a modified arduino, a homebrewed voltage regulator, an esp-01s (esp8266+1mb flash) and an rtc module....  all configurable through wifi with the mobile/pc (the relays will be added later)

this device for example take note of the last time it was active, and that info is stored in the eeprom (along with other configurations)....   i have another prototypes too, for bluetooth, sensors, etc ...  some of them work together

lcgamboa, 2021-12-18:
> so, that mean the option for backup/restore different projects (with firmwares) is "save/load workspace" ??? ... with flash (eeprom not yet) and all attached devices and cabling too?

Yes, the internal data eeprom memory is already saved in the workspace file. Only boards that have external eeprom memory that still don't work properly. 

In the case of arduino uno board  it should work without problems saving the workspace. 
To save the connections the spare parts window must be open when the workspace is saved (I want to change it to always save everything even with spare parts window closed).

I believe that a good improvement on PICSimLab is to show a dialog with an option to update the workspace file (which was used in loading) when it is finished, to remind the user.

lcgamboa, 2022-03-02:
> Only boards that have external eeprom memory that still don't work properly.

This was fixed in commit https://github.com/lcgamboa/picsimlab/commit/72aa2d80c4130beaa1cec57478557fc4a1703505

>I believe that a good improvement on PICSimLab is to show a dialog with an option to update the workspace file (which was used in loading) when it is finished, to remind the user.

This was implemented in commit https://github.com/lcgamboa/picsimlab/commit/371a7f21513d1b14ac54ca02f59c7aa043369b13

closed 2022-03-02 by lcgamboa