lcgamboa/picsimlab #54: Trying to program LCD1602 in PICGenios using PIC16F877A

eagl1, 2022-02-21:
Hi,

I'm working to get the LCD1602 embedded in PICGenios board using PIC16F877A.

I've tried this code but it's not working.

`
void pic_init(void){
    TRISA = 0b00000000;
    TRISB = 0b00000000;
    TRISD = 0b00000000;
    TRISC = 0b00000000;
    TRISE = 0b00000000;
}

void lcd_init(void){	
	__delay_ms(25);
    ADCON1 = 0b00001111;
    lcd_cmd(0x33);lcd_cmd(0x32);lcd_cmd(0x28);
    lcd_cmd(0x0f);lcd_cmd(0x01);lcd_cmd(0x80);	
}

void lcd_cmd(char cmd){
    PORTD = cmd;
    PORTE &= ~(1 << LCD_RS);       
    PORTE |= (1 << LCD_EN);
    __delay_us(10);
    PORTE &= ~(1 << LCD_EN);
    __delay_ms(5);
}

void lcd_data(char dat){
    PORTD =   dat; 
    PORTE |=  (1 << LCD_RS);
    PORTE |=  (1 << LCD_EN);
    __delay_us(10);
    PORTE &= ~(1 << LCD_EN);
    __delay_ms(5);
}

void move_cursor (unsigned char row, unsigned char col){
    if(row == 1){
        lcd_cmd(FIRST_ROW + col);
    }
        
    if(row == 2){
        lcd_cmd(SECOND_ROW + col);
    }
        
}  
    
void lcd_string_arr(char str[]){    
    char i;
    for(i=0; str[i] != 0; i++){
        lcd_data(str[i]);
    }
}

void lcd_string_ptr(const char *msg){
    unsigned char cnt=1;
	
    while((*msg) != 0){
        lcd_data(*msg);__delay_ms(2);
        msg++; cnt++;
        if(cnt == DDRAM_AFTER_CHAR16){
            lcd_cmd(SECOND_ROW);
            msg--;
        }
        if(cnt == DDRAM_AFTER_CHAR32){
            lcd_cmd(CLEAR_DISPLAY);
        } 
    }
}

void lcd_string_xy(unsigned char row, unsigned char col, char str[]){
    if(row == 1){
        lcd_cmd(FIRST_ROW + col);
    }
    if(row == 2){
        lcd_cmd(SECOND_ROW + col);
    }
	lcd_string_arr(str);
}`

In this main I'm using this test code:

`void main(void) {
    
    pic_init();
    lcd_init();    

    lcd_string_xy(0, 1, "lcd test");
    __delay_ms(2000);
    lcd_cmd(CLEAR_DISPLAY);     
    
    while(1){

    }

}`

lcgamboa, 2022-02-21:
Hi @eagl1 ,

Your code is for 8-bit mode and you are setting the display to 4-bit mode. 

```
void lcd_init(void){
__delay_ms(25);
ADCON1 = 0b00001111;
lcd_cmd(0x33); <-- normally the configuration must be done with sending three times the same value. 
lcd_cmd(0x32);
lcd_cmd(0x28);  <--- this line must be lcd_cmd(0x38) to work (or 0x33, 0x32)
lcd_cmd(0x0f);
lcd_cmd(0x01);
lcd_cmd(0x80);
}
``

eagl1, 2022-02-22:
Yes, also I found the source code here:
[https://github.com/lcgamboa/picsimlab_examples/tree/master/docs/board_McLab2/PIC16F877A/test_b3/src/test_b3.X](url)

And copied it, and it's working :)

Thanks.

closed 2022-02-22 by eagl1