lcgamboa/picsimlab #63: gdk pixbuf error

QiE2035, 2022-08-20:
```
Gtk-Message: 19:15:05.979: Failed to load module "canberra-gtk-module"
Gtk-Message: 19:15:05.979: Failed to load module "appmenu-gtk-module"
Gtk-Message: 19:15:06.007: Failed to load module "colorreload-gtk-module"
Gtk-Message: 19:15:06.007: Failed to load module "window-decorations-gtk-module"
Gtk-Message: 19:15:06.007: Failed to load module "appmenu-gtk-module"
PICSimLab: Load Config from file "/home/qie2035/.picsimlab/picsimlab.ini"
PICSimLab: Opening "/home/qie2035/.picsimlab/mdump_Blue_Pill_stm32f103c8t6.hex"
osc: 8000000
reset is called!

PICSimLab: qemu-stm32 -M stm32-f103c8-picsimlab-new -drive file=/home/qie2035/.picsimlab/mdump_Blue_Pill_stm32f103c8t6.bin,if=pflash,format=raw -d unimp -rtc clock=vm -icount shift=5,align=off,sleep=on                                                                                          
reset is called!

(picsimlab:162511): Gtk-WARNING **: 19:15:06.253: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/breeze/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/librsvg-2.so.2: undefined symbol: pango_coverage_get_type (gdk-pixbuf-error-quark, 5)
```

lcgamboa, 2022-08-20:
Are you using [PICSimLab-0.8.11_Linux_x86_64.AppImage](https://github.com/lcgamboa/picsimlab/releases/download/v0.8.11/PICSimLab-0.8.11_Linux_x86_64.AppImage)  version?
What is your Linux version?
Do you have the libgdk-pixbuf2.0-0 package installed on your system?
Seems to be related to an Appimage mime types issue: [Appimage fails if the pixbuf loaders/mime database missing](https://gitlab.com/inkscape/inkscape/-/issues/2969)

QiE2035, 2022-08-20:
Yes, 0.8.11 appimage x86_64, Arch Linx, pixbuf installed.

PICSimLab crash when I click the 'open bin' in File menu.

lcgamboa, 2022-08-20:
There are any error message in console when you open the .bin file?

QiE2035, 2022-08-20:
nop, just when i click that menu item, it crash, so I can't choose the .bin ile.

lcgamboa, 2022-08-20:
There really is a problem when the program tries to open a gtk filedialog using the Appimage.
 I was able to locate and reproduce the issue in the Appimage. 
Thanks for letting me know, I'll try to solve the problem.
As long as it is not fixed you can run your . bin using the command line: 
`./PICSimLab-0.8.11_Linux_x86_64.AppImage Blue_Pill stm32f103c8t6 file.bin`

QiE2035, 2022-08-20:
Not just open file, some operate also crash.

lcgamboa, 2022-08-20:
Can you try this[ Appimage](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/PICSimLab-0.8.11_220802_Linux_debian_sid_x86_64.AppImage) created in debian sid?

QiE2035, 2022-08-21:
Okay, not crash.
Maybe a small problem is the file dialog not use the GTK_USE_PORTAL=1 env var (I am using KDE)

lcgamboa, 2023-01-20:
Fixed in version [0.8.12](https://github.com/lcgamboa/picsimlab/releases/tag/v0.8.12) .

closed 2023-01-20 by lcgamboa