lcgamboa/picsimlab #64: PicsimLab will not run

synolope, 2022-08-27:
PicsimLab was working until I tried to change the board to STM32. Ever since I cannot run PicsimLab for the life of me. All the other applications that come with it works. I have tried reinstalling and restarting my PC. Nothing is working. If I try opening the program it just does not open. Please help.

synolope, 2022-08-27:
Note: If I try compiling it via the sh file I get this error:
`fatal error: lxrad.h: No such file or directory`

lcgamboa, 2022-08-27:
Hi @synolope ,
there are some instructions to use build scripts [here](https://lcgamboa.github.io/picsimlab_docs/stable/Installfromsource.html).
 Probably some problem with STM32 support is preventing the simulator from working, if you are using a Linux version open the program in a terminal and check the error messages.
It's possible to reset PICSimLab to the initial state changing the board with command line `picsimlab Arduino_Uno` or loading one of the [examples](https://lcgamboa.github.io/picsimlab_examples/board_Arduino_Uno.html).
Sending more information about which OS and PICSimLab version you are using can make it easier to detect the cause of the problem.

synolope, 2022-08-27:
i am running on windows 11 and picsimlab 0.8.11. i am installing it using the windows installer. resetting the board back to arduino uno worked! but, i really need the stm32 board. what can i do now?

synolope, 2022-08-27:
the following boards are not working for me: 
Blue Pill
ESP32
STM32

lcgamboa, 2022-08-27:
Looks like some problem with qemu support. 
The normal version for Windows does not generate messages in the terminal, please install the [experimental version](https://github.com/lcgamboa/picsimlab/releases/download/latestbuild/picsimlab_0.8.11_220802_win64_experimental_setup.exe) and check for any error messages in terminal.

serbianwizard, 2022-11-17:
> Hi @synolope , there are some instructions to use build scripts [here](https://lcgamboa.github.io/picsimlab_docs/stable/Installfromsource.html). Probably some problem with STM32 support is [...]

On Windows 10, running "picsimlab.exe Arduino_Uno" from the PicsimLab directory in Program Files via command prompt fixed this issue for me. Thanks!

lcgamboa, 2023-01-20:
The new version [0.8.12](https://github.com/lcgamboa/picsimlab/releases/tag/v0.8.12) has support to show error logs.

closed 2023-01-20 by lcgamboa