lcgamboa/picsimlab #67: I2C is not working on BluePill

byscg, 2022-10-31:
Hello, I tried to run the BMP280 module on the BluePill board, but I could not get the I2C communication to work. When I observed from the oscilloscope module, I saw that the I2C pins were 0V. Although I set the GPIO pins as pullup, I could not set them to 3.3V.

Does the BluePill board support I2C communication?

lcgamboa, 2022-10-31:
Hi @byscg, 
currently the port of qemu to stm32 (used by PICSimLab) does not support I2C and SPI as described on the official page [Qemu STM32](http://beckus.github.io/qemu_stm32/). 
In order to this work it is necessary to implement this support in qemu stm32, on the PICSimLab side the support is already ready and works for qemu ESP32.
It's on my to-do list, but since I don't have much time to work on the simulator and the qemu documentation doesn't help it will probably take some time to get implemented.
When I have something new, I'll contact you.

lcgamboa, 2022-11-26:
Hi @byscg,
  I added initial I2C support for STM32 in commit  https://github.com/lcgamboa/picsimlab/commit/e09e52b063999ca2e310208670ba3cf6f2e5b754 . You can test on the [latest version of build](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild).

lcgamboa, 2023-01-20:
Added in version [0.8.12](https://github.com/lcgamboa/picsimlab/releases/tag/v0.8.12) .

closed 2023-01-20 by lcgamboa