lcgamboa/picsimlab #69: Error compiling with Clang on macOS

fathonix, 2023-02-08:
I was trying to compile PICSimLab on my M1 MacBook Pro. After working around several hurdles, I eventually got things compiled (`picsim`, `lxrad`, `tty0tty` (but `pts` instead of `module`), `simavr`, `uCsim` and `qemu-stm32` ) except the PICSimLab itself.

Everything in `tools` like `PinViewer`, `espmsim` and `srtank` were successfully compiled, except `src`.  I got this error when running `make -j$(nproc)`:
```
/opt/homebrew/include/gpsim/errors.h:24:11: error: exception specification of overriding function is more lax than base version
  virtual ~TypeMismatch();
          ^
/opt/homebrew/include/gpsim/errors.h:8:7: note: overridden virtual function is here
class Error : public std::runtime_error {
      ^
```

But if I change `CXX` in Makefiles to GCC-12, everything was compiled but I got `tools/srtank` warnings and `src` errors linking to GPSim:
```
Linking srtank
warning: (arm64)  could not find object file symbol for symbol __ZN9CPWindow2D1Ev
warning: (arm64)  could not find object file symbol for symbol __ZN9CPWindow2D0Ev
warning: (arm64)  could not find object file symbol for symbol __Z5sopenPKc
warning: (arm64)  could not find object file symbol for symbol __Z6sclosev
warning: (arm64)  could not find object file symbol for symbol __Z5ssendh
warning: (arm64)  could not find object file symbol for symbol __Z4srecPh
warning: (arm64)  could not find object file symbol for symbol __Z9srec_toutPh
warning: (arm64)  could not find object file symbol for symbol __ZN9CPWindow2C2Ev
warning: (arm64)  could not find object file symbol for symbol __ZN9CPWindow2C1Ev
warning: (arm64)  could not find object file symbol for symbol __sub_I_65535_0.0
warning: (arm64)  could not find object file symbol for symbol __ZTS9CPWindow2
warning: (arm64)  could not find object file symbol for symbol __ZTI9CPWindow2
warning: (arm64)  could not find object file symbol for symbol __ZTV9CPWindow2
warning: (arm64)  failed to insert symbol 'GCC_except_table0' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table1' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table2' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table3' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table4' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table5' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table6' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table7' in the debug map.
warning: (arm64)  failed to insert symbol 'GCC_except_table8' in the debug map.
warning: (arm64)  could not find object file symbol for symbol _tw
warning: (arm64)  could not find object file symbol for symbol _wxTheAppInitializer
warning: (arm64)  could not find object file symbol for symbol __ZStL8__ioinit.lto_priv.0
warning: (arm64)  could not find object file symbol for symbol __ZStL8__ioinit.lto_priv.1
warning: (arm64)  could not find object file symbol for symbol __ZStL8__ioinit.lto_priv.2
```
```
Linking picsimlab
undef: __ZNKSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE7compareEmmPKc
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE6assignEPKcm
undef: __ZNSt3__113basic_ostreamIcNS_11char_traitsIcEEElsEx
undef: __ZTINSt3__115basic_streambufIcNS_11char_traitsIcEEEE
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE5imbueERKNS_6localeE
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE6setbufEPcl
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE7seekoffExNS_8ios_base7seekdirEj
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE7seekposENS_4fposI11__mbstate_tEEj
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE5uflowEv
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE9pbackfailEi
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE6insertEmPKcm
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE7replaceEmmPKc
undef: __ZNSt3__115basic_streambufIcNS_11char_traitsIcEEE6xsputnEPKcl
undef: __ZNSt3__113basic_ostreamIcNS_11char_traitsIcEEElsEm
undef: __ZNSt3__113basic_ostreamIcNS_11char_traitsIcEEElsEf
undef: __ZNSt3__1plIcNS_11char_traitsIcEENS_9allocatorIcEEEENS_12basic_stringIT_T0_T1_EEPKS6_RKS9_
undef: __ZNKSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE5rfindEcm
undef: __ZNKSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE7compareEPKc
undef: __ZNSt3__113basic_ostreamIcNS_11char_traitsIcEEElsEi
undef: __ZNSt3__18ios_base4initEPv
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE2atEm
undef: __ZNKSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE7compareEmmPKcm
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE9push_backEc
undef: __ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE6appendEPKcm
undef: __ZNSt3__113basic_ostreamIcNS_11char_traitsIcEEE3putEc
[... 411 lines cut ...]
```
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: arm64-apple-darwin21.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Output of `g++-12 -v`:
```
Using built-in specs.
COLLECT_GCC=g++-12
COLLECT_LTO_WRAPPER=/opt/homebrew/Cellar/gcc/12.2.0/bin/../libexec/gcc/aarch64-apple-darwin21/12/lto-wrapper
Target: aarch64-apple-darwin21
Configured with: ../configure --prefix=/opt/homebrew/opt/gcc --libdir=/opt/homebrew/opt/gcc/lib/gcc/current --disable-nls --enable-checking=release --with-gcc-major-version-only --enable-languages=c,c++,objc,obj-c++,fortran --program-suffix=-12 --with-gmp=/opt/homebrew/opt/gmp --with-mpfr=/opt/homebrew/opt/mpfr --with-mpc=/opt/homebrew/opt/libmpc --with-isl=/opt/homebrew/opt/isl --with-zstd=/opt/homebrew/opt/zstd --with-pkgversion='Homebrew GCC 12.2.0' --with-bugurl=https://github.com/Homebrew/homebrew-core/issues --with-system-zlib --build=aarch64-apple-darwin21 --with-sysroot=/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Homebrew GCC 12.2.0)
```

I use GPSim provided by Homebrew, which is compiled with `--disable-dependency-tracking`, `--disable-gui` and `--disable-shared` but I applied a patch from Debian to avoid `gpsim/breakpoints.h:25:10: fatal error: ../config.h: No such
file or directory`.

I'm myself not a C/C++ dev but familiar with building packages in Linux so I just tried running commands in `bscripts/build_all_and_install.sh` by hand and tweaking them to suit macOS.

Any help would be very appreciated. Thank you.

lcgamboa, 2023-02-08:
Thanks for your time and work.

From what I've read, c++17 renders this type of gpsim exception statement  as an error. Using c++14 it is just a warning. I believe that adding the `-std=c++14` option to clang++ the code should compile.

Using g++ should compile just fine, try removing the `-flto` option from makefiles, it's usually not very compatible.

fathonix, 2023-02-09:
Thanks for your feedback.

I've tried adding `-std=c++14` and the GPSim errors are finally gone. But I ran into errors like this:
```
error: value of type '__bind<int &, sockaddr *, unsigned long>' is not contextually convertible to 'bool'
```

But [quick search gave me the clue](https://stackoverflow.com/a/10035462) and I fixed it by replacing `bind()` with `::bind()` in devices/mplabxd.cc and devices/rcontrol.cc. Finally, PICSimLab was built successfully. (woohoo!)

[image]

I tried to flash some example programs to Uno simulator through pts-mode tty0tty and it worked like a charm. But once I switched to ESP32, I noticed that the serial had some issues connecting to /dev/ttys000. After confirming my configurations, PICSimLab hanged.

With arguments printed in the logs, I tried running:
```
./qemu-system-xtensa -M esp32-picsimlab -L /opt/lib/picsimlab/qemu/fw/ -drive file=/Users/aldo/Library/Application\ Support/picsimlab/mdump_ESP32_DevKitC_ESP32.bin,if=mtd,format=raw -icount shift=5,align=off,sleep=on -nic user,model=esp32_wifi,net=192.168.4.0/24,hostfwd=tcp::16555-192.168.4.15:80 -global driver=timer.esp32.timg,property=wdt_disable,value=true
```

Aaaand I got SEGV, here's the logs from macOS:
```
Process:               qemu-system-xtensa [49704]
Path:                  /Users/USER/*/qemu-system-xtensa
Identifier:            qemu-system-xtensa
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [39492]
Responsible:           iTerm2 [1389]
User ID:               501

Date/Time:             2023-02-09 21:38:36.8176 +0300
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        34581243-0F39-C375-0911-465CC4B97213

Sleep/Wake UUID:       A932D94F-0F3A-4720-A775-8B12DF635409

Time Awake Since Boot: 36000 seconds
Time Since Wake:       8659 seconds

System Integrity Protection: enabled

Crashed Thread:        3

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [49704]

VM Region Info: 0 is not in any region.  Bytes before following region: 4371185664
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1048b0000-1050f4000    [ 8464K] r-x/r-x SM=COW  ...system-xtensa

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a92d5598 __select + 8
1   libglib-2.0.0.dylib           	       0x106153090 g_poll + 424
2   qemu-system-xtensa            	       0x104f5eac4 qemu_poll_ns + 64 (qemu-timer.c:351)
3   qemu-system-xtensa            	       0x104f59e0c os_host_main_loop_wait + 88 (main-loop.c:257)
4   qemu-system-xtensa            	       0x104f59d1c main_loop_wait + 188 (main-loop.c:538)
5   qemu-system-xtensa            	       0x104ca2c30 qemu_main_loop + 32 (runstate.c:726)
6   qemu-system-xtensa            	       0x104d9e338 main + 56 (main.c:50)
7   dyld                          	       0x10606108c start + 520

Thread 1:
0   libsystem_kernel.dylib        	       0x1a92ce06c __semwait_signal + 8
1   libsystem_c.dylib             	       0x1a91d6fc8 nanosleep + 220
[... 90 lines cut ...]
MALLOC_MEDIUM (reserved)         440.0M        4         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.1M        5 
Stack                             10.1M        5 
VM_ALLOCATE                      130.5M       22 
VM_ALLOCATE (reserved)            24.3M       11         reserved VM address space (unallocated)
__AUTH                             220K       44 
__AUTH_CONST                      2892K      123 
__DATA                            4308K      126 
__DATA_CONST                      5778K      133 
__DATA_DIRTY                       233K       50 
__LINKEDIT                       586.6M       11 
__OBJC_CONST                       260K       29 
__OBJC_RO                         83.0M        1 
__OBJC_RW                         3152K        1 
__TEXT                            90.7M      139 
__UNICODE                          592K        1 
dyld private memory               1024K        1 
shared memory                      1.9G       18 
===========                     =======  ======= 
TOTAL                              3.9G      762 
TOTAL, minus reserved VM space     3.1G      762
```

I guess this is the one that caused the hang but I don't really understand what was going on.

lcgamboa, 2023-02-10:
Hi @fathonix , 
` qemu-system-xtensa -M esp32-picsimlab`  needs some initialization to work done by PICSimLab. I tweaked the code to work stand alone with this commit https://github.com/lcgamboa/qemu_stm32/commit/08ac82cc995683480333a9a73f66141e0826b4ca.
The problem of getting hang is because PICSimLab reads the status of the DTR/RTS pins of the serial port to reset the simulated boards for flashing the programs by avrdude and esptool.py. As you are using a pts without signals from these pins, the simulator enters a continuous reset state. To work with pts I can add an option to disable the DTR/RTS check, but for avrdude and esptoo.py to work it would be necessary to reset the boards manually and at the exact moment (I'm not sure if that would work).  For you to be able to do your tests just choose a real or invalid serial port for the simulator to run.
Thank you for your time and collaboration.

fathonix, 2023-02-10:
Thank you very much for your time and help.

I can confirm that `qemu-system-xtensa -M esp32-picsimlab` doesn't crash anymore (thanks!). I can also confirm that everytime I had flashed to Uno once, I had to reset the board either by selecting Arduino Bootloader or restarting the app sometimes, which I think it's because the IDE uses avrdude internally.

Regarding the serial, I'm still trying to find a virtual serial driver for macOS. Many people suggested [MacOSXVirtualSerialPort](https://github.com/clokey/PublicCode/tree/master/MacOSXVirtualSerialPort), which is very outdated and doesn't work on modern macOS. But I found [another driver](https://github.com/fracturedsoftware/Virtual-Serial-Port-for-OSX) and this one runs on newer macOS but still in early stage. The problem is that this driver isn't signed so it needs System Integrity Protection disabled, and I don't think that's the best idea. But signing requires Apple Developer membership and it costs $99 annually.

After fiddling around with Xcode build settings and macOS security stuff, apparently the driver is broken. I got it to run, I could connect on one side but the other side just threw "Resource busy." I don't put much hope on it either, since the dev kind of already abandoned it (the last commit is from 2018).

lcgamboa, 2023-02-11:
I'm going to put an option to be able to disable the use of DTR/DSR to work with the pts without getting hang. 
Another solution can be to use PICSimLab's TCP/IP rcontrol interface to load the programs. Here is a[ small program ](https://github.com/lcgamboa/picsimlab_rcontrol/tree/main/PLoader)that loads a .hex into an open instance of PICSimLab .

lcgamboa, 2023-02-11:
I have implemented the option to disable DTR/RTS signals use in commits https://github.com/lcgamboa/picsimlab/commit/573dc9cf4bcd642d468cc045ba0ffe0ec3f612cd and https://github.com/lcgamboa/picsimlab/commit/96f0a0965f8ded93d2b8bc9fb016e086c380219d.  Use the menu File->configure
[image]

With the option disabled, the reset must be done manually to flash the board.
To use with avrdude just reset at the beginning of the flashing attempt. 
To use with esptool.py, click on the BOOT button and drag the pointer out of the button leaving it pressed, click on the EN button and click on the BOOT button again, with that the ESP32 will be in boot mode.

laris, 2023-10-06:
Dear @fathonix,

I upgraded the macOS to Sonoma 14.0 and plan to build for native macOS binaries.
Could you share your Ventura build script and I can try to build to test?

Thanks, Laris

lcgamboa, 2023-10-06:
Hi @laris ,

there are some scripts in  [this pull request. ](https://github.com/lcgamboa/picsimlab/compare/master...hsorbo:picsimlab:hsorbo/macos-native)
I don´t know if it is the same used by @fathonix .

laris, 2023-10-06:
Thanks @lcgamboa, I struggle for that two scripts but still failed to compile full code.
I will try in next some days and recollect what failed and what needs to change.

laris, 2023-11-13:
@lcgamboa I'm trying to build dependence pkg one by one follow `build_all_static.sh`, when I try to build `picsimlab-stm32`, but the `build_libqemu-stm32.sh` will not generate `qemu-system-arm.rsp` file and also failed for below `sed` and `c++ -m64 -mcx16 -ggdb @qemu-system-arm.rsp`, is any missing steps?

```
flags="$(./configure --help | perl -ne 'print if s/^  ([a-z][\w-]*) .*/\1/' | tail -n +2 | awk '{print "--disable-"$1}' ORS=' ')"

${2:-.}/configure --target-list=$target --extra-cflags=-fPIC --disable-slirp $flags --enable-tcg \
        --enable-system --disable-werror --disable-alsa  \
        --enable-debug --enable-debug-info

make clean >/dev/null

# Build everything as usual
make "-j$ncpu"

# Build a shared library, without softmmu/main.o and otherwise *exactly* the same
# flags.
cd build
rm -f qemu-system-arm
ninja -d keeprsp
#dynamic
sed -i 's/qemu-system-arm.p\/softmmu_main.c.o//g' qemu-system-arm.rsp
sed -i 's/-o\ qemu-system-arm/-shared\ -o\ libqemu-stm32.so/g' qemu-system-arm.rsp
c++ -m64 -mcx16 -ggdb @qemu-system-arm.rsp
```

laris, 2023-11-13:
Yes, I understand the procedures (rm -f qemu-system-arm then ninja -d keeprsp) but it looks like the ninja command will don't generate the *.rsp file.

This is the configure parts (only run before make), could you help to double check if any package missing?
```ninja -vvv  -d keeprsp
[1/6] /usr/local/bin/meson --internal exe --capture qemu-version.h -- /Users/laris/coding/picsimlab/build_all/qemu/scripts/qemu-version.sh /Users/laris/coding/picsimlab/build_all/qemu '' 6.2.50
```
[qemu-stm32.log](https://github.com/lcgamboa/picsimlab/files/13342962/qemu-stm32.log)

lcgamboa, 2023-11-14:
Apparently there is no missing package/modules. 

Running ninja after remove qemu-system-arm inside build dir (in linux):
```
ninja -vvv  -d keeprsp
[1/5] /usr/bin/meson --internal exe --capture qemu-version.h -- /tmp/qemu/scripts/qemu-version.sh /tmp/qemu '' 6.2.50
[2/2] c++ -m64 -mcx16 @qemu-system-arm.rsp
```
The qemu-system-arm.rsp are generated.

Running without remove qemu-system-arm: 
```
[1/5] /usr/bin/meson --internal exe --capture qemu-version.h -- /tmp/qemu/scripts/qemu-version.sh /tmp/qemu '' 6.2.50
```
The qemu-system-arm.rsp aren't generated.
 
The .rsp file generate in Linux (for use with gcc), maybe it work with clang.
[qemu-system-arm.rsp.zip](https://github.com/lcgamboa/picsimlab/files/13343358/qemu-system-arm.rsp.zip)

laris, 2023-11-14:
I got one step forward:), need remove `qemu-system-arm qemu-system-arm-unsigned *unsigned` binary both, and ninja generate more codes but still don't generate *.rsp file.
```
mbp ~/c/p/b/q/build (picsimlab-stm32)> rm -rf qemu-system-arm qemu-system-arm-unsigned
mbp ~/c/p/b/q/build (picsimlab-stm32)> ninja -vvv  -d keeprsp
[1/6] /usr/local/bin/meson --internal exe --capture qemu-version.h -- /Users/laris/coding/picsimlab/build_all/qemu/scripts/qemu-version.sh /Users/laris/coding/picsimlab/build_all/qemu '' 6.2.50
[2/3] c++ -m64 -mcx16  -o qemu-system-arm-unsigned libcommon.fa.p/cpus-common.c.o libcommon.fa.p/page-vary-common.c.o libcommon.fa.p/disas_arm-a64.cc.o libcommon.fa.p/disas_libvixl_vixl_a64_decoder-a64.cc.o libcommon.fa.p/disas_libvixl_vixl_a64_disasm-a64.cc.o libcommon.fa.p/disas_libvixl_vixl_a64_instructions-a64.cc.o libcommon.fa.p/disas_libvixl_vixl_compiler-intrinsics.cc.o libcommon.fa.p/disas_libvixl_vixl_utils.cc.o libcommon.fa.p/disas_arm.c.o libcommon.fa.p/disas_i386.c.o libcommon.fa.p/disas_capstone.c.o libcommon.fa.p/hw_core_cpu-common.c.o libcommon.fa.p/hw_core_machine-smp.c.o libcommon.fa.p/qom_qom-hmp-cmds.c.o libcommon.fa.p/ui_clipboard.c.o libcommon.fa.p/ui_console.c.o libcommon.fa.p/ui_cursor.c.o libcommon.fa.p/ui_input-keymap.c.o libcommon.fa.p/ui_input-legacy.c.o libcommon.fa.p/ui_input-barrier.c.o libcommon.fa.p/ui_input.c.o libcommon.fa.p/ui_kbd-state.c.o libcommon.fa.p/ui_keymaps.c.o libcommon.fa.p/ui_qemu-pixman.c.o libcommon.fa.p/ui_util.c.o libcommon.fa.p/ui_spice-module.c.o libcommon.fa.p/ui_vnc-stubs.c.o libcommon.fa.p/audio_audio.c.o libcommon.fa.p/audio_audio_legacy.c.o libcommon.fa.p/audio_mixeng.c.o libcommon.fa.p/audio_noaudio.c.o libcommon.fa.p/audio_wavaudio.c.o libcommon.fa.p/audio_wavcapture.c.o libcommon.fa.p/chardev_msmouse.c.o libcommon.fa.p/chardev_wctablet.c.o libcommon.fa.p/chardev_testdev.c.o libcommon.fa.p/dump_dump-hmp-cmds.c.o libcommon.fa.p/block_blkreplay.c.o libcommon.fa.p/block_qapi-sysemu.c.o libcommon.fa.p/block_monitor_block-hmp-cmds.c.o libcommon.fa.p/softmmu_bootdevice.c.o libcommon.fa.p/softmmu_dma-helpers.c.o libcommon.fa.p/softmmu_qdev-monitor.c.o libcommon.fa.p/softmmu_device_tree.c.o libcommon.fa.p/backends_cryptodev-builtin.c.o libcommon.fa.p/backends_cryptodev.c.o libcommon.fa.p/backends_hostmem-ram.c.o libcommon.fa.p/backends_hostmem.c.o libcommon.fa.p/backends_rng-builtin.c.o libcommon.fa.p/backends_rng-egd.c.o libcommon.fa.p/backends_rng.c.o libcommon.fa.p/backends_confidential-guest-support.c.o libcommon.fa.p/backends_rng-random.c.o libcommon.fa.p/backends_hostmem-file.c.o libcommon.fa.p/backends_cryptodev-vhost.c.o libcommon.fa.p/migration_page_cache.c.o libcommon.fa.p/migration_xbzrle.c.o libcommon.fa.p/migration_vmstate-types.c.o libcommon.fa.p/migration_vmstate.c.o libcommon.fa.p/migration_qemu-file-channel.c.o libcommon.fa.p/migration_qemu-file.c.o libcommon.fa.p/migration_yank_functions.c.o libcommon.fa.p/migration_block-dirty-bitmap.c.o libcommon.fa.p/migration_channel.c.o libcommon.fa.p/migration_colo-failover.c.o libcommon.fa.p/migration_colo.c.o libcommon.fa.p/migration_exec.c.o libcommon.fa.p/migration_fd.c.o libcommon.fa.p/migration_global_state.c.o libcommon.fa.p/migration_migration.c.o libcommon.fa.p/migration_multifd.c.o libcommon.fa.p/migration_multifd-zlib.c.o libcommon.fa.p/migration_postcopy-ram.c.o libcommon.fa.p/migration_savevm.c.o libcommon.fa.p/migration_socket.c.o libcommon.fa.p/migration_tls.c.o libcommon.fa.p/migration_block.c.o libcommon.fa.p/monitor_hmp-cmds.c.o libcommon.fa.p/monitor_hmp.c.o libcommon.fa.p/monitor_qmp-cmds.c.o libcommon.fa.p/net_announce.c.o libcommon.fa.p/net_checksum.c.o libcommon.fa.p/net_colo-compare.c.o libcommon.fa.p/net_colo.c.o libcommon.fa.p/net_dump.c.o libcommon.fa.p/net_eth.c.o libcommon.fa.p/net_filter-buffer.c.o libcommon.fa.p/net_filter-mirror.c.o libcommon.fa.p/net_filter-rewriter.c.o libcommon.fa.p/net_filter.c.o libcommon.fa.p/net_hub.c.o libcommon.fa.p/net_net.c.o libcommon.fa.p/net_queue.c.o libcommon.fa.p/net_socket.c.o libcommon.fa.p/net_util.c.o libcommon.fa.p/net_filter-replay.c.o libcommon.fa.p/net_tap-bsd.c.o libcommon.fa.p/net_tap.c.o libcommon.fa.p/net_can_can_core.c.o libcommon.fa.p/net_can_can_host.c.o libcommon.fa.p/replay_replay.c.o libcommon.fa.p/replay_replay-internal.c.o libcommon.fa.p/replay_replay-events.c.o libcommon.fa.p/replay_replay-time.c.o libcommon.fa.p/replay_replay-input.c.o libcommon.fa.p/replay_replay-char.c.o libcommon.fa.p/replay_replay-snapshot.c.o libcommon.fa.p/replay_replay-net.c.o libcommon.fa.p/replay_replay-audio.c.o libcommon.fa.p/replay_replay-random.c.o libcommon.fa.p/replay_replay-debugging.c.o libcommon.fa.p/hw_acpi_acpi_interface.c.o libcommon.fa.p/hw_acpi_aml-build.c.o libcommon.fa.p/hw_acpi_bios-linker-loader.c.o libcommon.fa.p/hw_acpi_core.c.o libcommon.fa.p/hw_acpi_utils.c.o libcommon.fa.p/hw_acpi_acpi-cpu-hotplug-stub.c.o libcommon.fa.p/hw_acpi_memory_hotplug.c.o libcommon.fa.p/hw_acpi_nvdimm.c.o libcommon.fa.p/hw_acpi_pci.c.o libcommon.fa.p/hw_acpi_generic_event_device.c.o libcommon.fa.p/hw_acpi_ghes.c.o libcommon.fa.p/hw_acpi_acpi-pci-hotplug-stub.c.o libcommon.fa.p/hw_acpi_viot.c.o libcommon.fa.p/hw_acpi_erst.c.o libcommon.fa.p/hw_acpi_ipmi-stub.c.o libcommon.fa.p/hw_acpi_acpi-x86-stub.c.o libcommon.fa.p/hw_adc_stm32f2xx_adc.c.o libcommon.fa.p/hw_adc_aspeed_adc.c.o libcommon.fa.p/hw_adc_npcm7xx_adc.c.o libcommon.fa.p/hw_adc_zynq-xadc.c.o libcommon.fa.p/hw_adc_max111x.c.o libcommon.fa.p/hw_audio_soundhw.c.o lib [... cut ...] p.fa @block.syms @qemu.syms -lz /usr/local/Cellar/pixman/0.42.2/lib/libpixman-1.dylib @block.syms -lutil -lm -L/usr/local/Cellar/glib/2.78.1/lib -L/usr/local/opt/gettext/lib -lgthread-2.0 -lglib-2.0 -lintl -lz -framework CoreFoundation -framework IOKit -lc++
ld: warning: -undefined error is deprecated
ld: warning: ignoring duplicate libraries: '-lc++', '-lz'
[3/3] /Users/laris/coding/picsimlab/build_all/qemu/scripts/entitlement.sh qemu-system-arm qemu-system-arm-unsigned ../pc-bios/qemu.rsrc
```

laris, 2023-11-14:
repeat to test:
1. in build path `rm -rf qemu-system-arm qemu-system-arm-unsigned`
2. `make clean`
3. `ninja -d keeprsp` then will generated `qemu-system-arm qemu-system-arm-unsigned` file but no `qemu-system-arm.rsp`

```
ninja -d keeprsp
[820/2547] Compiling C object libqemuutil.a.p/meson-generated_.._trace_trace-hw_ppc.c.o
[1095/2547] Linking static target libqemuutil.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_cacheflush.c.o) has n
o symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_stats64.c.o) has no s
ymbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_int128.c.o) has no sy
mbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(stubs_module-opts.c.o) has
 no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_cacheflush.c.o) has n
o symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_stats64.c.o) has no s
ymbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(util_int128.c.o) has no sy
mbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libqemuutil.a(stubs_module-opts.c.o) has
 no symbols
[1240/2547] Compiling C object libblock.fa.p/block_file-posix.c.o
../block/file-posix.c:3330:18: warning: 'IOMasterPort' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
    kernResult = IOMasterPort( MACH_PORT_NULL, &masterPort );
                 ^~~~~~~~~~~~
                 kIOMainPortDefault
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Head
ers/IOKitLib.h:143:1: note: 'IOMasterPort' has been explicitly marked deprecated here
IOMasterPort( mach_port_t       bootstrapPort,
^
1 warning generated.
[1268/2547] Linking static target libchardev.fa
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libchardev.fa(chardev_char-parallel.c.o)
 has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libchardev.fa(chardev_char-parallel.c.o)
 has no symbols
[1436/2547] Compiling C object libcommon.fa.p/hw_char_stm32_uart.c.o
../hw/char/stm32_uart.c:230:13: warning: explicitly assigning value of variable of type 'uint64_t' (aka 'unsigned long long') to itself [-Wse
lf-assign]
  curr_time = curr_time; // Avoid "variable unused" compiler error
  ~~~~~~~~~ ^ ~~~~~~~~~
1 warning generated.
[1458/2547] Compiling C object libcommon.fa.p/hw_display_sii9022.c.o
../hw/display/sii9022.c:69:13: warning: enumeration value 'I2C_START_SEND_ASYNC' not handled in switch [-Wswitch]
    switch (event) {
            ^~~~~
../hw/display/sii9022.c:69:13: note: add missing switch cases
    switch (event) {
            ^
1 warning generated.
[1459/2547] Compiling C object libcommon.fa.p/hw_display_ssd0303.c.o
../hw/display/ssd0303.c:190:13: warning: enumeration value 'I2C_START_SEND_ASYNC' not handled in switch [-Wswitch]
    switch (event) {
            ^~~~~
../hw/display/ssd0303.c:190:13: note: add missing switch cases
    switch (event) {
            ^
1 warning generated.
[1497/2547] Compiling C object libcommon.fa.p/hw_i2c_smbus_slave.c.o
../hw/i2c/smbus_slave.c:67:13: warning: enumeration value 'I2C_START_SEND_ASYNC' not handled in switch [-Wswitch]
    switch (event) {
            ^~~~~
../hw/i2c/smbus_slave.c:67:13: note: add missing switch cases
    switch (event) {
            ^
1 warning generated.
[... 266 lines cut ...]
[2498/2547] Linking target tests/qtest/fuzz-sdcard-test
ld: warning: -undefined error is deprecated
[2502/2547] Linking target tests/qtest/microbit-test
ld: warning: -undefined error is deprecated
[2503/2547] Linking target tests/qtest/cdrom-test
ld: warning: -undefined error is deprecated
[2504/2547] Linking target tests/qtest/arm-cpu-features
ld: warning: -undefined error is deprecated
[2505/2547] Linking target tests/qtest/machine-none-test
ld: warning: -undefined error is deprecated
[2510/2547] Linking target tests/qtest/device-introspect-test
ld: warning: -undefined error is deprecated
[2511/2547] Linking target tests/qtest/test-arm-mptimer
ld: warning: -undefined error is deprecated
[2515/2547] Linking target tests/qtest/qmp-test
ld: warning: -undefined error is deprecated
[2516/2547] Linking target tests/qtest/qmp-cmd-test
ld: warning: -undefined error is deprecated
[2517/2547] Linking target tests/qtest/qom-test
ld: warning: -undefined error is deprecated
[2518/2547] Linking target tests/qtest/test-hmp
ld: warning: -undefined error is deprecated
[2547/2547] Linking target tests/qtest/qos-test
ld: warning: -undefined error is deprecated
```

laris, 2023-11-14:
I tried to reduce the backed simulator but still cannot compile picsimlab.
provide some files as reference.
- Makefile.Common, reduce backed
- Makefile.macos.static, build Makefile
- build cmdline ` make CXX="ccache g++ <+option for gnu++11 or c++14>" -j16 LIBPATH="../build_all/" FILE=Makefile.macos.static 2>&1 | tee log-file-name`
- `make-stdout-stderr-output.txt`, `-std=gnu++11`
- `make-stdcpp14-stdout-stderr-output.txt` ` -std=c++14`
[Makefile.Common.txt](https://github.com/lcgamboa/picsimlab/files/13346057/Makefile.Common.txt)
[Makefile.macos.static.txt](https://github.com/lcgamboa/picsimlab/files/13346058/Makefile.macos.static.txt)
[make-stdout-stderr-output.txt](https://github.com/lcgamboa/picsimlab/files/13346059/make-stdout-stderr-output.txt)
[make-stdcpp14-stdout-stderr-output.txt](https://github.com/lcgamboa/picsimlab/files/13346098/make-stdcpp14-stdout-stderr-output.txt)
```
$ pwd
picsimlab/src
$ make CXX="ccache g++ -std=gnu++11" -j16 LIBPATH="../build_all/" FILE=Makefile.macos.static 2>&1 | tee make-stdout-stderr-output.txt
$ make CXX="ccache g++ -std=c++14" -j16 LIBPATH="../build_all/" FILE=Makefile.macos.static 2>&1 | tee make-stdcpp14-stdout-stderr-output.txt

```

laris, 2023-11-14:
I almost pass and only one error now.

```
override CXXFLAGS+= -D_LIBCPP_HAS_NO_OFF_T_FUNCTIONS
override CXXFLAGS+= -D_DARWIN_C_SOURCE
```

```
error: value of type '__bind<int &, sockaddr *, unsigned long>' is not contextually convertible to 'bool'
        if (bind(listenfd, (sockaddr*)&serv, sizeof(serv))) {
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

lcgamboa, 2023-11-14:
Try to replace:
`if (bind(listenfd, (sockaddr*)&serv, sizeof(serv))) {`
with:
`if (bind(listenfd, (sockaddr*)&serv, sizeof(serv)) != 0) {`

laris, 2023-11-14:
I update two bind to `(::bind(listenfd, (sockaddr*)&serv, sizeof(serv)))` and fixed.

laris, 2023-11-14:
I got SEGFAULT :)
Maybe need other steps
```
PICSimLab: Version "0.9.1 231112 DARWIN native"
PICSimLab: Command Line: ./picsimlab --help 
PICSimLab Caught SIGSEGV: Segmentation Fault
```

laris, 2023-11-14:
looks like works now after add `OBJS+= boards/bridge_gpsim.o`

laris, 2023-11-15:
[Is it possible to separate the backend simulator/emulator when compile the full package? · Issue #98 · lcgamboa/picsimlab](https://github.com/lcgamboa/picsimlab/issues/98)

https://github.com/laris/picsimlab

forked version, only provide simavr backend and AVR boards

OBJS+= boards/bsim_simavr.o # simavr backend
OBJS+= boards/board_Arduino_Mega.o
OBJS+= boards/board_Arduino_Nano.o
OBJS+= boards/board_Arduino_Uno.o
OBJS+= boards/board_Franzininho_DIY.o

closed 2023-12-07 by lcgamboa