lcgamboa/picsimlab #73: LEDs with push button don't hold up on continuous press

eagl1, 2023-03-22:
Hi,

I have a problem when testing simple Arduino code to activate LEDs with push button press.

This is the code:

`
void setup() {
  pinMode(0, OUTPUT);
  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, INPUT);
  
}

void loop() {
  if(digitalRead(8)){
    digitalWrite(0, HIGH);   
    digitalWrite(1, HIGH);   
    digitalWrite(2, HIGH);   
    digitalWrite(3, HIGH);   
    digitalWrite(4, HIGH);   
    digitalWrite(5, HIGH);   
    digitalWrite(6, HIGH);   
    digitalWrite(7, HIGH);       
  }
  else{
    digitalWrite(0, LOW);    
    digitalWrite(1, LOW);    
    digitalWrite(2, LOW);    
    digitalWrite(3, LOW);    
    digitalWrite(4, LOW);    
    digitalWrite(5, LOW);  
    digitalWrite(6, LOW);    
    digitalWrite(7, LOW);     
  }
}
`

[image]

The problem is that LEDs don't hold up ON state with continuous push button press. It blinks and go back to LOW state.

lcgamboa, 2023-03-23:
Hi @eagl1 ,

 your code is correct.
 If you're trying to use the PinViewer's toggle button, it only works if you don't have any other connections on the same pin.

[image]

If you connect more than one signal source to one pin, the simulator will not work correctly. For example, put two pushbuttons on the same pin or try to use PinViewer to trigger a pin that already has a signal source.

eagl1, 2023-03-23:
If I use PinViewer's alone, it works OK.

But when using push buttons or switches, it doesn't work.

Here's the basic test using only switches and LEDs, so when the switch is in ON state, the output blinks ON for a moment then returns to OFF.

[image]

closed 2023-04-15 by lcgamboa