lcgamboa/picsimlab #74: Load Cell & HX711 Module

RyanGee00, 2023-03-22:
Hi, can you please add Load Cell & HX711 Module configuration into PICSIMLAB? As I would lke to link the load cell to my project for further reference.
[image]

lcgamboa, 2023-03-23:
Hi @RyanGee00 ,

I've never used a Load Cell and an HX711. But from the datasheet it seems to be not very difficult to implement. I'll put it on the to-do list, and as soon as I have time I'll try to implement it.

lcgamboa, 2023-03-27:
Hi @RyanGee00 ,
 
HX711 part added in the last unstable release https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild

[image]

RyanGee00, 2023-03-27:
Thank you very much for that kind & quick responses!

closed 2023-04-15 by lcgamboa