lcgamboa/picsimlab #78: Error loading saved Workspace on Windows

peterm02019, 2023-04-13:
Hi,
 Thank you very much for all the work which has obviously gone into this software project. It looks good so far (I'm a new user), but I'm having a couple of issues with it and I hope that we can get these sorted out so I can begin using it with confidence.

I am getting this error (see attached screenshot) when trying to Load a previously saved Workspace in PICSimLab. I think I see the problem, at least on the surface. I think that Windows is bound to fail trying to find a path with forward slashes in it, but perhaps this may not be the whole story? 

[image]

Of course, now that I try it again from scratch, the error does not occur! I'll leave it here anyway in case it provides a clue as to something which might come back to bite later.

Thanks very much & have a great day!

lcgamboa, 2023-04-14:
Hi @peterm02019 ,

slashes in the file path are not a problem for the wxwidigets used by PICSimLab. 
The workspace file is a zipped file. When PICSimLab opens the pzw file it is unzipped inside the temp folder. I believe that your problem can be caused by some antivirus checking the file and making it unavailable at the time of reading. I can't reproduce the problem on my computer to try to eliminate it.
Do you use any kind of antivirus? Can you reproduce the problem? Any tips and more information are welcome.
Thank you for contacting us.

closed 2023-06-03 by lcgamboa

peterm02019, 2023-06-10:
Thank you Luis.

I have not been able to get this problem to happen again, and I'm sorry for
not getting back to you. Too much in action here.

Thank you for taking care of it. Sorry for a late reply. For some unknown
reason, some of my emails arriving are confused by my normal program.

All the best,
Peter