lcgamboa/picsimlab #80: ESP32 DevKitC ethernet example

JanuNoo, 2023-05-08:
Hello,
I have read that simulating  ethernet  on ESP32 DevKitC board is supported. However I am having problems simulating a simple  ethernet web server.
After I try to begin ethernet using "Ethernet.begin" in arduino the code runs but the IP address returned by "Ethernet.localIP()" is "0.0.0.0".
When simulating, I have checked option "Enable Ethernet (-nic user,model=open_eth,net=192.168.3.0/24)" but got no luck.
Also I have tried adding ETH W5500 part (from Modules->Spare Parts-> ETH W5500) but also no luck.
Maybe, if it is not to much trouble, you could include an example of a minimal ethernet server running in a ESP32 DevKitC board?.
If not, maybe you can explain to me what I am doing wrong and how to fix it?.
Thanks.

lcgamboa, 2023-05-09:
Hi @JanuNoo ,
So far I've only tested with opencores support in IDF and it worked without problems. I believe because it is not a real driver, opencores does not have support turned on in the Arduino.
https://github.com/espressif/qemu/wiki#enabling-ethernet-support
Theoretically, the ESP32 should work with the ETH W5500 connected to the SPI port, but I haven't done that test yet.
Below the examples:  https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_request and https://github.com/espressif/esp-idf/tree/master/examples/protocols/http2_request configured to run on PICSimLab:

[esp32_eth_test.zip](https://github.com/lcgamboa/picsimlab/files/11426123/esp32_eth_test.zip)

It's only work on the last version of PICSimLab https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild

JanuNoo, 2023-05-09:
Hello,
I can confirm that the new examples you posted work indeed.
Regarding on arduino:
I tried using ESP32 with the ETH W5500 connected to the SPI port but again no luck.
I have tried manually configuring the CS pin ("Ethernet.init(10);") and without.
Because W5500 part does not support DHCP I used a static IP address ("Ethernet.begin(mac, ip)").
For configuring W5500 part I used the following pins:

W5500 	<---> 	ESP32
MOSI 	<---> 	GPIO23
MISO 	<---> 	GPIO19
SCK 	        <---> 	GPIO18
SS/CS 	<---> 	GPIO5
INT 	        <---> 	GPIO4

Also I tried adding this one too:
RST 	        <---> 	EN (pin 2)

I don't know if I am doing something wrong or if it is just not possible.
Also, does "picsimlab" support SPIFFS? if not, do you have any plans to support it in the future?
Thanks

lcgamboa, 2023-05-11:
I confirm that a code that works for the real w5500 and  ESP32 does not work in the simulator. 
I detected some issues that need to be fixed. The main thing is that the data written by the SPI is not being generated correctly in MOSI pin. The current implementation of the w5500 also does not yet implement the interrupt output and the version register (which is not documented in the datasheet most used by the idf driver).

The W5500 implemented in PICSimLab has DHCP support implemented, but the W5500 IP address is not used for packet routing, the localhost address is always used.

Yes, the qemu-esp32 (used by PICSimLab) has support to SPIFFS.

As soon as I have time I will try to find the source of the problem. Any news I'll get in touch.

JanuNoo, 2023-05-11:
Hello,
Thanks for the fast and useful replies. Everything is more clear for me now.
And thanks also for this wonderful project, is very nice.

lcgamboa, 2023-05-14:
I solved the qemu-esp32 SPI problem. But I couldn't make it work with the IDF ethernet support because it uses the MACRAW mode of the w5500 that is not yet implemented in PICSimLab. I believe it is possible to implement MACRAW mode using RAW sockets (which usually requires administrator privileges), but I won't have time to implement that for now. And I believe it will take a considerable amount of work.

closed 2023-07-29 by lcgamboa