lcgamboa/picsimlab #81: PinViewer crash

JanuNoo, 2023-05-30:
Hello,
I have tried to simulate gpio input events with ESP32 DevKitC board and when I toggled the pin I wanted to test with PinViewer it just crashed.
I did the following:

//define queue
gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
//start gpio task
xTaskCreate(gpio_task, "gpio_task", 2048, NULL, 10, NULL);

//set pin for input
 int INPUT_PIN =18; //defined GPIO 18 for input
 //zero-initialize the config structure.
 gpio_config_t io_conf = {};
 //interrupt of rising edge
 io_conf.intr_type = GPIO_INTR_POSEDGE;
 uint64_t pin_bit_mask=((1ULL<<INPUT_PIN));
 io_conf.pin_bit_mask = pin_bit_mask;
 //set as input mode
 io_conf.mode = GPIO_MODE_INPUT;
 //enable pull-up mode
 io_conf.pull_up_en = 1;
 gpio_config(&io_conf);

 //add pin listener
 gpio_install_isr_service(0);
 gpio_isr_handler_add(INPUT_PIN, gpio_isr_handler, (void*) INPUT_PIN);

static void IRAM_ATTR gpio_isr_handler(void* arg)
 {
 uint32_t gpio_num = (uint32_t) arg;
 xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
 }

static void gpio_task(void* arg)
 {
 uint32_t io_num;
 for(;;) 
  {
  if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) 
   {
   ESP_LOGI(TAG, "GPIO[%"PRIu32"] intr, val: %d\n", io_num, gpio_get_level(io_num));
   }
  }
 }

From the error log file I got:

PICSimLab: Debug On=0  Type=MDB Port=1234
PICSimLab: Remote Control Port 5000
Spare parts: parts[00] (LCD ili9341) created 
**
ERROR:../accel/tcg/tcg-accel-ops.c:81:tcg_handle_interrupt: assertion failed: (qemu_mutex_iothread_locked())
PICSimLab Error: EXCEPTION_BREAKPOINT
PICSimLab stack: 00007ffd74e4b3b2
PICSimLab stack: 00000000685daf6a
PICSimLab stack: 00000000685dcd9d
PICSimLab stack: 00000000685dd224
PICSimLab stack: 00007ffcdfa32284
PICSimLab stack: 00007ffcdf78f3f8
PICSimLab stack: 00007ffcdf80bb71
PICSimLab stack: 00007ffcdf80bd01
PICSimLab stack: 00007ffcdfa3e1ab
PICSimLab stack: 00007ffcdf812ab3
PICSimLab stack: 00007ffcdfa3e1ab
PICSimLab stack: 00007ffcdf641f23
PICSimLab stack: 00007ffcdfa3e1ab
PICSimLab stack: 00007ffcdf81356c
PICSimLab stack: 00007ffcdf8138cb
PICSimLab stack: 000000000045230c
PICSimLab stack: 0000000000401a2b
PICSimLab stack: 00007ffd2f0cd802
PICSimLab stack: 00007ffd3650de4d
PICSimLab stack: 00007ffd3650df9f
PICSimLab stack: 00007ffd772caf5a
PICSimLab stack: 00007ffd772cb02c
PICSimLab stack: 00007ffd76e47614
PICSimLab stack: 00007ffd775226a1

Any ideas of what the problem is?
Thanks

JanuNoo, 2023-05-31:
Hi,
I tested again using push buttons as you said and I confirm that works indeed.
Thanks

lcgamboa, 2023-06-02:
Hi @JanuNoo,

the problem with PinViewer as fixed in commit a646624 . Try the [latest build](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild) version with the fix .

JanuNoo, 2023-06-05:
Works, thanks.

closed 2023-06-05 by JanuNoo