lcgamboa/picsimlab #82: The application freeze and exit on gpboard

eagl1, 2023-06-04:
Hi,

I'm just trying other boards, and I wanted to try the gpboard, but when I selected debug, the application got freeze and exited.

[image]

lcgamboa, 2023-06-05:
Hi @eagl1,

the problem that you detected is not because of the debug, but because this specific microcontroller does not have all the configuration words implemented in the gpsim simulator. It was fixed in commit 847292f .

The gpboard doesn't have debug support implemented yet.  Consider using breadboard board that supports PIC18F4550 and debug.

Thank you for reporting the problem.

eagl1, 2023-06-05:
Do you mean I can update the simulator ?

closed 2023-06-12 by lcgamboa