lcgamboa/picsimlab #83: Support RISC-V MCUs simulation

tpimh, 2023-06-12:
I would like to simulate a RISC-V MCU, for example, SiFive FE310. What are the available options? Possibly, QEMU can be used similar to ARM and Xtensa?

lcgamboa, 2023-06-12:
Hi @tpimh ,

I recently added support for a board with the ESP32-C3 which is a 32 bits RISC-V  90bfddd.   This board uses espressif's modified version of [qemu.](https://github.com/lcgamboa/qemu)
As support for C3 peripherals is not yet complete, several things still don't work.
It is possible to add other processors supported by qemu, just adapt some peripherals to communicate with PICSimLab. The sifive_u machine is already supported by qemu-system-riscv32 used by ESP32-C3. By making similar modifications it would be possible to add other processors.

tpimh, 2023-06-15:
It seems that all required code is already in QEMU, all I need is `qemu-system-riscv32 -machine sifive_e`.

closed 2023-07-29 by lcgamboa