lcgamboa/picsimlab #84: Hangs with com0com

thomasn1975, 2023-06-30:
**Describe the bug**
When used with com0com and the terminal at the other virtual COM port is not connected, SICSimLab crashes.

**To Reproduce**
In my case COM5 and COM6 are set as connected virtual COM ports. PICSimLab is set to use COM6.
Load a HEX that periodically sends data via Serial.write().
As long as a listening terminal is connected to COM5, everything is working fine.
Whenever the terminal is not connected (at startup or while running the simulation), the speed indicator decreases and the simulation hangs. Soon after, PICSimLab is not responding anymore.

I do not think this is a problem with com0com, as a command line tool - that is working exactly the same - works well.

**Desktop (please complete the following information):**
 - OS: Win10 64

lcgamboa, 2023-06-30:
Hi @thomasn1975 ,

have you checked the “enable buffer overrun”  com0com option for both or one virtual port?

thomasn1975, 2023-06-30:
Ummm... no, I did not... and now I have found the documentation for that.

Sorry for the inconvenience.

closed 2023-06-30 by thomasn1975