lcgamboa/picsimlab #85: Support for PIC16F819,PIC16F1827 and PIC16F1847

djsbriscoe, 2023-07-14:
Hello,
I would like to help the project by contributing support for the following PIC microcontrollers
PIC16F819
PIC16F1827
PIC16F1847
How can I do this? What changes need to be made to which files, or would new files need to be created? If so, could you provide specific details of what work would need to be done? Could pre-existing files that are similar be edited to reduce the amount of work required? I have some experience of C programming but limited experience of C++. However, I am willing to use any contribution I could make as a learning opportunity. Thank you.

lcgamboa, 2023-07-14:
Hi @djsbriscoe ,

Support for PICs microcontrollers should be added primarily in another project, the [PICsim](https://github.com/lcgamboa/picsim).

Support for a new microcontroller is done by adding a file with the description of its registers in the [src/pics ](https://github.com/lcgamboa/picsim/tree/master/src/pics)folder.

Adding new microcontrollers is relatively simple, I have some automated scripts that help with the task. The bigger job is to test and see if the peripherals are working properly.

djsbriscoe, 2023-07-14:
Hi,
Thanks for your reply. If you are able to create the file for the PIC16F819 (and for the PIC16F1827 and the PIC16F1847 if you have the time) in the src/pics folder named above then I might have a go at testing them. Or maybe send the files and then I can compile a local version of PicSimLab in order to do some tests. What do you recommend as the best way to do these tests? I use the PIC16F819 a lot as we teach an embedded systems class using it here in the UK. I hope you can help. Thanks.

lcgamboa, 2023-07-15:
Hi @djsbriscoe,

I've added support for the [microcontrollers you requested](https://github.com/lcgamboa/picsimlab/releases/tag/latestbuild). I just did a simple blink and debug test with mplabx. The best way to test is to use codes that have already been validated on real hardware in the simulator and check if the behavior is the same.

djsbriscoe, 2023-07-15:
Hi,
I'm working on XC8 code for the PIC16F819 (the other 2 later on) at the moment. I will run code that passes my hardware tests on the simulator as I get a chance. The tests on the hardware takes precedence, but it would be really nice if it can also work on the simulator as well, hence this request. Thank you for your time in doing this.

djsbriscoe, 2023-07-16:
Do I have to download and compile the picsimlab source code in order to use this? Thanks.

djsbriscoe, 2023-07-16:
Hi,
Is it possible for you to produce a demo/reference project for the PIC16F819 (a simple LED Blink if a switch pressed for example? When I open a breadboard the PIC16F819 is not centred on the breadboard (it's positioned off centre) and then when I open a spare parts window and try to select led's and switches NOTHING appears in the spare parts window. I'm using Windows 10 Pro V22H2 Build 19045.3208. Thanks.

lcgamboa, 2023-07-16:
Hi @djsbriscoe ,

the two projects you requested are attached in the projects.zip file (in the end).  In the spare parts window, after selecting from the menu, it is necessary to click on the position on the screen where the part must be inserted, look at the first image.

[image]

[image]

PICSimLab has a different interface from most simulators, any suggestion to try to improve it for the user is welcome.

[projects.zip](https://github.com/lcgamboa/picsimlab/files/12065193/projects.zip)

closed 2023-07-29 by lcgamboa