lcgamboa/picsimlab #87: PICSimLab closing or crashing

omolayoabegunde, 2023-08-11:
Good day sir, I have issue with PICSimlab. I am using Fedora 38 precisely, anytime I launch and select on configure or load hex file from the file menu, the picsimlab will close or crash. I have try several means. Below are error am getting. Pls help me

PICSimLab: Version "0.9.0 230729 Linux64_WX Appimage"
PICSimLab: Command Line: /tmp/.mount_PICSimXpQkHK/usr/bin/picsimlab 
PICSimLab: Load Config from file "/home/rapture/.picsimlab/picsimlab.ini"
PICSimLab: Window position x=256 y=141
PICSimLab: Window scale  1.00 
PICSimLab: Using board "Arduino Uno"
PICSimLab: Instance number 0
PICSimLab: Opening "/home/rapture/.picsimlab/mdump_Arduino_Uno_atmega328p.hex"
PICSimLab: Error on Port Open /dev/tnt2! 
/dev/tnt2: No such file or directory
PICSimLab: Debug On=0  Type=MDB Port=1234
PICSimLab: Remote Control Port 5000
PICSimLab: Saving "/home/rapture/.picsimlab/mdump_Arduino_Uno_atmega328p.hex"
PICSimLab: End Board Simulation.
PICSimLab: Load Config from file "/home/rapture/.picsimlab/picsimlab.ini"
PICSimLab: Window position x=256 y=141
PICSimLab: Using board "Arduino Uno"
PICSimLab: Instance number 0
PICSimLab: Opening "/home/rapture/.picsimlab/mdump_Arduino_Uno_atmega328p.hex"
PICSimLab: Error on Port Open /dev/tnt2! 
/dev/tnt2: No such file or directory
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
PICSimLab: Debug On=0  Type=MDB Port=1234
PICSimLab: Remote Control Port 5000
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
PICSimLab Caught SIGABRT: usually caused by an abort() or assert()
PICSimLab Stack[00]: /lib64/libc.so.6(gsignal+0x1e) [0x7f12eda5fabe]
PICSimLab Stack[01]: /lib64/libc.so.6(abort+0xdf) [0x7f12eda4887f]
PICSimLab Stack[02]: /tmp/.mount_PICSimXpQkHK/usr/lib/libglib-2.0.so.0(+0x1db43) [0x7f12ede81b43]
PICSimLab Stack[03]: /tmp/.mount_PICSimXpQkHK/usr/lib/libglib-2.0.so.0(g_assertion_message_error+0xb6)

lcgamboa, 2023-08-11:
Hi @omolayoabegunde,

this is a problem related to appimage. Have you tested both available .appimage versions? One works on older Linuxes and others on newer ones.

omolayoabegunde, 2023-08-12:
Yes sir! I have tried other image versions, but I might not know the actual version you are talking about. I have not tried it on older Fedora version, as I stated earlier sir, I am using Fedora 38 (new  one)

lcgamboa, 2023-08-13:
This is based in ubuntu  stable:
 https://github.com/lcgamboa/picsimlab/releases/download/v0.9.0/PICSimLab-0.9.0_230729_Linux_x86_64.AppImage 
and this is based on Debian Test:
 https://github.com/lcgamboa/picsimlab/releases/download/v0.9.0/PICSimLab-0.9.0_230729_Linux_Debian_Test_x86_64.AppImage.

closed 2023-08-14 by omolayoabegunde