lcgamboa/picsimlab #89: custom svg rendering issue

fariouche, 2023-08-20:
Hello,

I want to add a new board, the m5stack core2.
I've designed the svg with inskape, and created the map with gimp.
The resulting files are attached.
My issue is that the bottom part of the svg is not rendered properly, and the map file does not map properly the middle screen too...

I'm starting to understand how it works, and I have made some changes to add support for psram and 16MB of flash.
I have an other issue, but it will be on an other ticket.

Thanks!
[image]

fariouche, 2023-08-20:
The map file, I can't upload it so I embed it here:
`<img src="[M5stackCore2] (imported)" width="425" height="425" border="0" usemap="#map" />
<map name="map">

<area shape="circle" coords="16,100,10" href="I_SW_PWR" />
<area shape="circle" coords="324,408,13" href="I_SW_RST" />
<area shape="circle" coords="280,407,12" href="O_LD_GREEN" />
<area shape="rect" coords="8,210,26,258" href="B_USB_PORT" />
<area shape="rect" coords="63,97,357,319" href="B_DISPLAY" />
</map>
`

lcgamboa, 2023-08-20:
Hi @fariouche ,

your map file lacks the first line with the size of the picture. This size is used by PICSimLab.

```
<img src="[picsimlab7] (overwritten)" width="425" height="425" border="0" usemap="#map" />

<map name="map">
<!-- #$-:Image map file created by GIMP Image Map plug-in -->
<!-- #$-:GIMP Image Map plug-in by Maurits Rijk -->
<!-- #$-:Please do not edit lines starting with "#$" -->
<!-- #$VERSION:2.3 -->
<!-- #$AUTHOR:fariouche  -->
<area shape="circle" coords="18,102,8" href="I_SW_PWR" />
<area shape="circle" coords="324,407,11" href="I_SW_RST" />
<area shape="circle" coords="280,408,10" href="O_LD_GREEN" />
<area shape="rect" coords="11,211,26,256" href="B_USB_PORT" />
<area shape="rect" coords="62,99,359,320" href="B_DISPLAY" />
</map>
```

fariouche, 2023-08-20:
Strange, it is what gimp outputed ... I followed your doc. 
Maybe I missed something? Or the svg is also wrong and missing the size too?

lcgamboa, 2023-08-20:
Your SVG it's ok.
Add the line:
`<img src="[picsimlab7] (overwritten)" width="425" height="425" border="0" usemap="#map" />` 
to your map file to make it work.
[image]

fariouche, 2023-08-21:
The svg is missing the bottom part (a led and the reset button)
The same kind of button to the left(USB c and power button) are fine.
thanks for the map, I will try

fbenamrouche, 2023-08-21:
Just had a look at my map file and discovered that I already have the line... I missed it when I copied it here...
Anyway, I've found that the map size didn't match the svg size for some reason. FIxed that and it now works, thanks!

fariouche, 2023-08-21:
I can continue now rhanks

closed 2023-08-21 by fariouche