lcgamboa/picsimlab #90: edp32 devkitc MHz settings issues

fariouche, 2023-08-20:
Hello,

When playing with the edp32 devkitc board, I see that the MHz are set to 1.... I tried to set it to 40MHz, and the emulator is running way slower!
Is it expected behavior?

Thanks

lcgamboa, 2023-08-20:
Hi again @fariouche ,

For boards that use Qemu as a simulator, the frequency combobox updates only the hardware update frequency of PICSimLab, the speed of Qemu is controlled by the `CPU Mips` combobox.
Increasing the PICSimLab update consumes more resources and usually slows down Qemu's simulation.
Increasing the speed usually improves the accuracy of asynchronous signals, such as those used in reading a DHT22 sensor. For synchronous signals like I2C or SPI it makes no difference.

fariouche, 2023-08-21:
I'm not sure I understand "hardware update frequency"... You mean the spare parts are running using an independent clock from the semi one and are only updated using this clock?

lcgamboa, 2023-08-21:
In Qemu the cpu runs at the maximum frequency that the host allows and the qemu peripherals are updated using timers synchronized in the system wall time. With the icont option activated, the qemu cpu tries to synchronize with the wall time, but without being cycle acurrated. PICSimLab is updated with every Qemu IO operation or by a Qemu timer for synchronization every 100ms. If the PICSimLab frequency is 1MHz it will run 100k steps every 100ms, if it is 10MHz it will run 1M steps every 100 ms.

fariouche, 2023-08-21:
I understand! So one tick per millisecond at 1mhz is already a lot! I understand why it slows down now.

Thanks!

closed 2023-08-21 by fariouche