lcgamboa/picsimlab #91: [feature request] add pause button

fariouche, 2023-08-20:
Hello again,

This time it's not a bug, but sometimes I would like to have a pause button for example to have time to read the log output etc

Is it difficult to add?

Thanks

lcgamboa, 2023-08-20:
Hi @fariouche ,

The idea of PICSimLab is to simulate a real board, I don't think a pause button directly on the interface is a good idea.
Using the [Remote Control Interface ](https://lcgamboa.github.io/picsimlab_docs/stable/RemoteControlInterface.html) you can use the `sim stop` and `sim start` commands to pause and resume the simulation.

fariouche, 2023-08-20:
I was more thinking about a button inside one of the menu... Like the load program entry

Or better, since we can change the emulator speed, a speed of zero for example.

Anyway, it's not a big problem. I will see with the remote interface, didn't tried that yet

closed 2023-08-21 by fariouche