lcgamboa/picsimlab #92: issue loading esp32 firmware

fariouche, 2023-08-22:
Hello,

I'm getting problems loading my firmware. I've setup the application offset variable accordingly, but still no boot.
It is working with the default devkitC example.

After some more investigations I've found two problems.
One is that only 4MB flash is support while my app was built for 16MB flash. I had to comment the code that check the mdump file size, and added a new variable to set the flash size.

The second issue is that the generated flash file does not contain all the partitions.
I've built a script that must be run from inside the build folder of esp idf that extract automatically all the info and create a full image including the bootloader etc.
I do have built also a platformio script that do the same. 
No Arduino ide support for the moment.

I will push a PR when ready.
But I need first to clean stuff and add proper testing. Not sure yet about the best way to do it

fariouche, 2023-08-23:
Yes this is ecact for the second problem.
For the first, I didn't see that function, I will have a look. But the issue is that the esp qemu does set the flash size depending the image size. And from what I understood, picsimlab search for a mdump, if none, takes the one in the demo workspace and copy at application_offset into mdump. So no bootloader etc here, it seems that the full image prepared here is incomplete.
I didn't try the load image menu yet

lcgamboa, 2023-08-24:
Yes, in the latest version of PICSimLab, the demo project is loaded when there is no mdump file. In the case of your new board, it would be enough to create a demo file with 16M. Demo (.pzw) and mdump always have the size of flash and usually include the partitions and bootloader. If an empty mdump file is loaded, only flashing with esptool will work. 
Loading one file with a size smaller than Flash (using menu)  will use the `application_offset`, but without bootloader simulation will not work (as in real board). 

You can override the method:
`unsigned int DBGGetROMSize(void) override{ return 16*1024*1024}; `
to define the size of Flash of your new board. In the next weekend I will add support to use DBGGetROMSize in bsim_qemu class instead 4M fixed size.

fariouche, 2023-08-26:
Thanks!
I have two other questions, not related to this problem...
1. Picsimlab exposes pins of the devkitC. In some situations, some MCU pins are not exposed but used by internal devices like a LCD display. I see that the qemu picsimlab machine is exporting only the gpio is of the devkitC. So I exported all the gpio, but then they appear all in the picsimlab interface. If I map only the exposed gpios, using the gitpinname méthode etc, then I cannot use any other gpios. Anyway this is not a big deal, so I export all the esp32 gpios regardless of the board. What is the intended behavior of picsimlab? Only expose the board pins, or all the possible gpios even if they are not routed?

2. I can receive I2C data from the running program using the bitbang I2C Io functions. From what I see, every time a SDA gpio changes state, run CPU is called, and I2C Io function rebuild the data from the bits. What confuses me is that the I2C ctrl api exist and seems to be used by the qemu. The behavior is different, as it seems that qemu expects a full byte of data... The callback in picsimlab that handles this read I2C request starts a timer and tries to reconstruct the data from the bits... But the returned  data to qemu is zero here as the qemu callback expects the read data now and not after some time from a timer. Not sure I'm clear 😅 

I will be home soon, and I will have access to my computer to continue properly and start creating a fork.

lcgamboa, 2023-08-26:
Hi @fariouche ,

1- You are correct, it is necessary to refactor the code for the pin mapping to be done in PICSimLab instead of Qemu. I think it's better to just expose the board's pins to the user (As it happens on a real board.).

2-There really is a hack to make this work. As you said, Qemu writes and waits to read the I2C transaction at once. When PICSimLab receives an I2C byte, it synchronizes the simulation time and transmits the bits in a future time in relation to Qemu. This is not the actual behavior of an I2C hardware and the Qemu CPU hangs while the I2C transaction is timing ahead. I haven't found a better way to make it work. (In the latest version of Qemu it seems that support for an I2C transaction callback has been introduced)

lcgamboa, 2023-08-26:
Custom Flash size using DBGGetROMSize(void) implemented in commit 09fdccd

fariouche, 2023-08-26:
Great, I will have a look 

For the I2C, the bitbang I2C Io seems to work... That means that qemu is properly changing the I2C gpio states when transferring.
I don't know for the timings, but at least no special hack is needed here. And I believe that the pic simulator works also that way

lcgamboa, 2023-08-26:
The ESP32 Qemu doesn´t change gpio states and doesn´t implement the bitbang interface for I2C, UART or SPI, this interface is implemented in PICSimLab (in the `bsim_qemu ` class).

fariouche, 2023-08-27:
This is also what I believed... until I tried to called bitbang_i2c_io(&rtc->bb_i2c, pins[22], pins[23]); from run_CPU_ns()...
(in my case my program uses gpios 24 and 25, converted to pin numbering (gpio - 1)...) and I was able to get the relevant data (correct i2c address and register for my device). I was surprised as I didn't expect it to work too...
However, I didn't try to send data yet... maybe this side will not work. I'm not using master_i2c[id]

lcgamboa, 2023-08-27:
The commit efff3a0 moves the pin mapping from the Qemu machine to PICSimLab boards. 
Now it is possible to change pins in new boards like yours.

fariouche, 2023-08-27:
Wow! Thank you!
One last question, still related to gpio. My board uses an axp192. It's a power manager ic over I2C. One of its feature is gpio expander. 4 gpio can be controlled via I2C. It also have 4 ldo with variable voltage. Some of the gpio and ldo are connected to the LCD. There is also a battery charger.

They are not exposed to the pin header behind, only used internally.
I know that this can simulated by spare parts, but in my case, it's easy to add the app and LCD directly in the board file. 
I can add an array to hold the axp gpio states, but what about the exposed axp pins?
I was thinking about putting them after the qemu pins because it is big enough.

Later, I will try to add a li-ion battery as a spare part.

lcgamboa, 2023-08-27:
You can implement the axp192 and the display as "devices" and use them on your board and as spare parts.
 As spare parts you can use the VCD dump part to save the signals, it is handy to find errors using pulseview to analyze and decode protocols such as I2C.
You can use pins with values greater than the pin count returned by the [MGetPinCount](https://github.com/lcgamboa/picsimlab/blob/efff3a0c6743a96084eb0ef3b2a512e1789382be/src/boards/board_DevKitC.cc#L1206C21-L1206C33) method and less than the initial value declared by the [IOINIT](https://github.com/lcgamboa/picsimlab/blob/efff3a0c6743a96084eb0ef3b2a512e1789382be/src/lib/spareparts.h#L31C9-L31C15) spare parts window (currently 110).
You can also add pins to be exposed by increasing the value returned by the `MGetPinCount` method, as is done on the [PQBB board ](https://github.com/lcgamboa/picsimlab/blob/efff3a0c6743a96084eb0ef3b2a512e1789382be/src/boards/board_PQDB.cc#L1588) that adds the outputs of a shift register to the list of pins.

fariouche, 2023-08-28:
Just done a fresh clone of the repo, recompiled everything.... (no modifications from my side) and picsimlab when first run launch the Curiosity board... then I switch to DevKitC and nothing.... nothing in the error log.... relaunching picsimlab, I get an error popup warning, I switch back to the DevKitC and I see the board and qemu running, but does nothing (ie, spare parts window is empty and no blinking led)...
I'm investigating what is wrong

fariouche, 2023-08-29:
The root cause is indeed because of the mdump file in demo.pzw that does not contain a bootloader. 
I merged my board and with my program it works thanks

fariouche, 2023-08-30:
still have some strange behavior when loading from fresh....
I remove ~/.picsimlab, execute picsimlab. I have the Curiosity board that is started. I select the DevKitC board... and nothing.
Using gdb, I see that the code enters bsim_qemu, then tries to open fname_ that is ~/.picimlab/mdump_DevKitC_ESP32.bin.... and does not found it, so it creates an empty one. (at this stage, the /tmp folder contains the workspace from the pzw file and is correct, but ~/.picsimlab does not have any of the workspace files yet, they are created later)...
If I then remove all the ~/.picsimlab/*DevKitC* files and relaunch picsimlab, they are properly created now and everything runs fine

fariouche, 2023-08-30:
So, when I load the board from the menu (only the first time as explained above), I see that it quits. This is the backtrace I see just before this happens:
#0  qmp_quit (errp=0x0) at ../monitor/qmp-cmds.c:46
#1  0x0000555555678583 in bsim_qemu::MEnd (this=0x555555fd8870) at boards/bsim_qemu.cc:909
#2  0x00005555557726f3 in CPICSimLab::EndSimulation(int, char const*) [clone .constprop.0] () at lib/picsimlab.cc:408
#3  0x000055555576b9eb in CPICSimLab::LoadWorkspace(wxString, int) [clone .constprop.0] () at lib/picsimlab.cc:527
#4  0x000055555576977c in CPICSimLab::Configure(char const*, int, int, char const*, int) [clone .constprop.0] ()
    at lib/picsimlab.cc:1225
#5  0x0000555555650545 in menu1_EvBoard (this=0x0, control=0x7fffffffcc30)
    at /home/fariouche/apps/src/iot/picsimlab/src/picsimlab1.cc:1108
#6  0x00007ffff729c666 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
    () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#7  0x00007ffff729c9da in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#8  0x00007ffff729ca70 in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#9  0x00007ffff729cb2b in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#10 0x00007ffff729cc01 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#11 0x00007ffff781026a in wxWindowBase::TryAfter(wxEvent&) () from /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#12 0x00007ffff729c967 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#13 0x00007ffff77c360a in wxMenuBase::SendEvent(int, int) () from /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0

It is loading the workspace and then Ends the simulation.... in bsim_qemu. I can understand that it ends the simimulation because we are switching to a new board, but the current board is Curiosity, not qemu based... so the qemu simulation already started, then quickly ended to load the workspace.... then I don't know yet why it is not resuming. (maybe some error, it is exiting because of exit(0) called in picsimlab.cc line 485, leaving the workspace corrupted)

lcgamboa, 2023-08-30:
I have fixed some issues when mdump size is not the expected flash size.  Try with the changes of this new commit e078d75 . With the blank mdump file, Qemu freezes randomly.

fariouche, 2023-08-30:
still exactly the same behavior :-(
This only happens when starting picsimlab from fresh (no .picsimlab folder) and then switching to qemu board... picsimlab exits ... and the reason is because GetNeedReboot() is true.
And if I remove the devkitc files from the picsimlab home folder, they are re-created fine.
I do not see this behavior with my other slower laptop. My PC is fast, looks like a timing issue

fariouche, 2023-08-31:
From what I can see... when i click on the button to switch board, I see that the qemu emulator is started and then stopped right away, then the simulation should restart, but does not... Most likely a threading conflict issue...

lcgamboa, 2023-08-31:
I have done several tests. Apparently it's a problem with Qemu's threads. When the Qemu ESP32 is loaded with an empty Flash file, it sometimes freezes (on my computer). I've only been able to reproduce your problem on an older computer I have, and even then it only happens one out of twenty attempts. With Qemu ESP32-C3 (which only has one core) which has basically the same implementation in PICSimLab the problem does not occur. Could you test with the ESP32-C3-DevKitC-02 board and see if the problem occurs on your computer?

fariouche, 2023-08-31:
Unfortunately, same problem with the C3 board :(
I will continue to debug this week end. 
When I switch the board, what should happen? I imagine that the current simulation (Curiosity) is ended, and a new qemu one is started. But why does is try to set the needReboot flag?
I tried to comment the SetNeedReboot in the bsim_qemu() constructor, and it crashes in qemu_init (because it is already initialized, but why? This is the first time I start the qemu emulator)

lcgamboa, 2023-09-01:
At first, PICSimLab ended the simulators (simavr, PICsim and ucsim) and restarted (the same or another simulator). For this there is support from these simulators to finalize and "clean" their context. For Qemu this is not possible, Qemu has no simulation finalization support. As it was developed to be an application, the OS does the cleaning when it is finished.
So when the simulator used by PICSimLab is Qemu (or gpsim) the needReboot flag is set and when the simulation is finished (board change) the PICSimLab application is closed and another instance is opened. Running the first time without any mdump file, to switch from the curiosity board (PICSim) to the DevKitC (Qemu) the sequence is:
1-Curiosity board is closed
2-The DevKitC board is opened and an empty mdump file is created
3-It is detected that it is the first time and flagged to load the demo
4-DevKitC board is closed (PICsimLab is closed)
5-The new instance of PICSimLab opens loading the demo.

I currently use to develop a Ryzen 7 4800H processor running Linux Debian Testing.
Without being able to reproduce the problem it is difficult to try to find the cause.
 What configuration and OS are you using?

fariouche, 2023-09-01:
I'm using a Ryzen 9 3900X 12-Core, 64GB of ram and brtfs raid10 ssd, under gentoo.
Yes, not easy when we cannot reproduce. But I can reproduce it 100% of the time. Now that I understand how the thing should work, I can but breakpoints and check, no worries. I will keep you posted

fariouche, 2023-09-01:
ok, nothing to do with argv.... it's just that the lxGetDisplayWidth() gets an error and abort the execution, don't know why.

fariouche, 2023-09-02:
Finally figured out the root cause of the bug, nasty one!
It is crashing in X11 lib because the socket file descriptor used by xcb is 1... it was most likely recycled because it was closed before the fork. and the child thinks stdout is still fd 1, closing it, and making the whole xcb communication failing.

To test this, I commented the line "freopen(fname, "w", stdout);" in picsimlab1.cc:761 and now it works (and picsimlab sends all its output to the terminal now of course)

The question now is why does the child think that stdout fd is 1? (because even if the parent does close stdout, the child should assign a new stdout fd if needed)

lcgamboa, 2023-09-02:
This error doesn't make any sense in my point of view.
Theoretically, the new child process would be independent of the parent process. It could be a problem with the lxExecute() function (which for lxradWX is the wxWidgets wxExecute() function). Try passing the second parameter with the flags ` lxEXEC_ASYNC | lxEXEC_MAKE_GROUP_LEADER`  int the [line](https://github.com/lcgamboa/picsimlab/blob/e078d7557e7640012e117b3a8fe5ce0661de5b57/src/lib/picsimlab.cc#L484C3-L484C3) .
The line should look like: ` lxExecute(cmd,lxEXEC_ASYNC | lxEXEC_MAKE_GROUP_LEADER );` 
Check if there are any differences in execution.
For testing you can also replace the `lxExecute(cmd)` function with the `system(cmd)` function and check if it works.

You can turn off stdout redirection by adding the line below in the Makefile:
`override CXXFLAGS+= -D_CONSOLE_LOG_`

fariouche, 2023-09-02:
The child is independant, but inherits all the file descriptors of the parent at the moment of fork. So if a fd is closed before the fork, it will be closed in the child. Of course, after the fork the child is independant of the parent and the file descriptors are independant too.

My theory is that the code that does the freopen() in picsimlab1.cc and the code that loads the boards are run in different threads... the code that handles the wxwidget calls opens a socket with fd 1 (because it is available), but in parallel, it is freopened() (I do not know if freopen is atomic...)
I will make the test, but I'm almost sure that lxExecute with ASYNC (that is the default) and GROUP_LEADER will not change anything. the system() call should be identical, will try too.
_CONSOLE_LOG_ will work for sure.

fariouche, 2023-09-02:
some more info. The bug is only triggered when I delete de full .picsimlab folder.
Here is what happens exactly:
- When starting the first time, freopen() is called, but fname (the picsimlab_log0.log file) does not exist.... the function closes the stdout file descriptor. (I can see it in /proc/<pid>/fd that fd 1 does no exist anymore after the call)
- Later, the lxExecute() is called. This creates the child, that inherits no fd 1.
- The child open the socket to x11, and is assigned fd 1.
- The child call freopen()... that closes fd 1 and open the log file. I checked the stdout FILE* structure, and the file descriptor is 1, even if the current fd for stdout is not 1.

fariouche, 2023-09-02:
That just reminds me of something old... and done in systemd if I remember well.
"in the child, loop from 0 to getrlimit (RLIMIT_NOFILE,...) - 1 and if FD_ISSET for that file descriptor
in the fd_set for this particular child role, close it"
I just realize that the children always assume (in the glibc) that stdout is 1 and open! (same for stderr and stdin)

lcgamboa, 2023-09-02:
Did using the flags in the lxExecute() function or the system() function make any difference?

The appimage run without this error in Gentoo? 

This link of Stack Overflow   [Is it possible that `fileno(stdout) != 1` on a POSIX system?](https://stackoverflow.com/questions/25516375/is-it-possible-that-filenostdout-1-on-a-posix-system) say that is possible stdout be different of one. But the code example always returns 1 on my system.  

It could be something specific to Gentoo, brtfs (I use ext4), or the versions of the libraries used. I am still unable to reproduce this very strange problem.

fariouche, 2023-09-02:
using the flags in the lxExecute() function or the system() makes no difference.

Yes, having fileno(stdout) different of 1 is possible. But here the interesting point (that have no answer yet) is what happens to the FILE* of stdout if the stdout is closed by the parent before the fork...
The thing here is that this only happens when the log file does not exist. If I create an empty one, it works.

At least this makes perfect sense now to me... freopen() closes stdout because the log file does not exist, then fork, then the child connect to x11 and gets allocated fd 1 (I've checked, it's in main(), very early, we wx is initializing)... and finally freopen() is called using FILE* stdout that has _fileno still set to 1 (at least in my case), gets closed and reopened to the fd of the log file.... and now the x11 communication is broken.

There is no threading here.... it should be 100% reproducible everywhere, not only on my computer.
Maybe on your system, calling freopen() with an non existing file does not close the stdout (different version of glibc may have different behavior... or it is linked to the filesystem itself as you suggest)

Can you check on your computer what happens if freopen() is called when no picsimlab home folder?

On my side:
just before freopen() (before the fork):
```
fariouche@Quazar ~/apps/src/iot/picsimlab $ ls -l /proc/14261/fd
total 0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 0 -> /dev/pts/0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 1 -> /dev/pts/0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 10 -> 'socket:[7492302]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 11 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:32 12 -> 'socket:[7484864]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 2 -> /dev/pts/0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 3 -> 'socket:[7496045]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 4 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 5 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 6 -> 'socket:[7501262]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 7 -> 'anon_inode:[eventfd]'
lr-x------ 1 fariouche fariouche 64 Sep  2 16:29 8 -> anon_inode:inotify
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 9 -> 'anon_inode:[eventfd]'
```
And just after freopen() (as you can see, fd 1 disapeared):
```
fariouche@Quazar ~/apps/src/iot/picsimlab $ ls -l /proc/14261/fd
total 0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 0 -> /dev/pts/0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 10 -> 'socket:[7492302]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 11 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:32 12 -> 'socket:[7484864]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 2 -> /dev/pts/0
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 3 -> 'socket:[7496045]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 4 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 5 -> 'anon_inode:[eventfd]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 6 -> 'socket:[7501262]'
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 7 -> 'anon_inode:[eventfd]'
lr-x------ 1 fariouche fariouche 64 Sep  2 16:29 8 -> anon_inode:inotify
lrwx------ 1 fariouche fariouche 64 Sep  2 16:29 9 -> 'anon_inode:[eventfd]'
```

lcgamboa, 2023-09-02:
Here the file descriptors 1 and 2 are mapped to /dev/null as expected.  
Debian Testing uses libc6 version 2.37-7.

[image]

lcgamboa, 2023-09-02:
Hi @fariouche ,

I found the problem and fixed it in commit 0a29f26. 
You were right. I was doing the wrong test just deleting the files, when I deleted the directory the error appeared. 
The problem is that freopen cannot create the file if the directory in the path does not exist. And this causes the file descriptor 1 to be closed and used by another fopen function, which causes the error at fork time.
Thanks for the help, sorry for the delay in the solution.

fariouche, 2023-09-03:
That's good news! No worries, I'm glad to help!
Your fix works! thanks.
I can now continue implementing the device :)

closed 2023-09-03 by fariouche